gnuradio/gnuradio #4732: Control of message delay

GuilleA97, 2021-06-06:
Hi, 
I'm trying to send some messages at the same time using blocks 'Message Strobe', but I need control the timing when the message are sending. 'Message Strobe Random Delay' don't the solution because I need choose the time, no sent it in a random valor of time.

Thank u.

willcode, 2021-06-06:
Message Strobe allows you to choose the period, and the period can also be controlled by a variable or widget, e.g., a text entry or slider.

Please ask for assistance on the mailing list or chat.gnuradio.org. If you have a feature request, please reopen this issue and provide a more detailed explanation of the feature.

closed 2021-06-06 by willcode

willcode, 2021-06-06:
Do you mean that you would like to generate multiple different messages simultaneously and send them to different places, and have them arrive at the same time?

jacobagilbert, 2021-06-06:
@GuilleA97 This is something I have considered before. I am midway through upstreaming a number of PDU/MSG blocks with some time based features.

Can you submit a feature request to the mailing list like @willcode suggested and explain your use case in detail?

willcode, 2021-06-06:
I wonder if having an internal tag-to-message-handler thing in basic_block would address this kind of thing. The two concepts of time (stream time and system time) make simultaneous stuff hard, and delivery doesn't really happen at the same time no matter what you do.

jacobagilbert, 2021-06-06:
What time base to use makes it complicated, and messages are asynchronous by design, but I have definitely run into situations where these can be worked around into something useful. Any amplifying use cases that are reasonably extensible help support the case to implement something here in tree.