gnuradio/gnuradio #4736: Master branch fails build on Ubuntu 18.04 due to log4cpp

mormj, 2021-06-07:
The log4cpp in ubuntu 18.04 is super old (1.1.1), and cannot handle the C++17 requirement that was introduced for GR 3.10.

Really nothing we can do about it other than another motivation to get rid of log4cpp (latest release was 2017)

mormj, 2021-06-08:
Not inherently a GR issue, though - just added it here as a note, so closing.

closed 2021-06-08 by mormj

jackdondy, 2021-09-30:
@mormj , I just came across with the same problem you mentioned above and wonder how you deal with it.
I am installing GR 3.9.2.0, end up with:
```
/usr/include/log4cpp/Priority.hh:107:2: error: ISO C++1z does not allow dynamic exception specifications
  throw(std::invalid_argument);
```
and several similar errors.