gnuradio/gnuradio #4742: string == "0.0" isn't sufficient for testing zeroness of string

marcusmueller, 2021-06-08:
Soapy Python bindings do this:

https://github.com/gnuradio/gnuradio/blob/f29be5398081e78ed0edeaf712cc608e770f2523/gr-soapy/python/soapy/bindings/soapy_common.cc#L60-L66

this will be hilarious if someone tries 0e0, -0, or +0 and gets something different than if they use 0.0.

Let's not have that much hilarity in our lives :D

willcode, 2021-06-08:
I remember leaving that one for latter ... and forgetting. Thanks for catching it.

willcode, 2021-06-08:
@ncorgan what do you think, since this was copied from the 0.8.0 code?

marcusmueller, 2021-06-08:
I illustrate what I'd recommend in #4743, together with upstreaming these changes in https://github.com/pothosware/SoapySDR/pull/313 to keep things consisten.

PR #4743 (2021-06-11): soapy: less surprising bool cast
Soapy has its own way of casting strings to bools, and it special-cases
"0" and "0.0" and misses out on the opportunity to be consistent with
many other ways of specifying 0 as number.

I'm trying to upstream the same change to SoapSDR, in order to keep
things consistent. But this is worth having in GNU Radio, either way,
since GNU Radio has its own user expectations, which might even differ a
bit from these of SoapySDR users.

Fixes #4742

References https://github.com/pothosware/SoapySDR/pull/313
closed 2021-06-11 by willcode