gnuradio/gnuradio #4757: Python invalid syntax- inserting extra comma

suciokhan, 2021-06-11:
Building low-pass-filter through the GUI results in this (note the extra comma after 75000 which throws a Python error:
```
self.low_pass_filter_0 = filter.fir_filter_ccf(
            20,
            firdes.low_pass(
                1,
                4000000,
                75000,
                ,
                firdes.WIN_HAMMING,
                6.76))
```
Here is what the error looks like:
[image]

Here is my low-pass-filter config in the GUI:
[image]

Note: Every time I execute it, the Transition Width gets blanked-out.

willcode, 2021-06-11:
What version of gnuradio?

Does this happen with a new flowgraph?

suciokhan, 2021-06-11:
'3.8.1.0'

Hmmm...after closing down gnuradio_commpanion, I manually fixed the Python script using vim, and now when I open it up I can hit play and it works.

willcode, 2021-06-11:
Good, closing. Probably a stale value that got stuck in a grc file after a gnuradio update.

closed 2021-06-11 by willcode