gnuradio/gnuradio #4759: corr_est block finds no correlations when "Symbols" argument is purely complex

sodaBitters, 2021-06-11:
I'm generating a PN sequence and modulating it BPSK (DSSS) in python, then casting to complex64 and exporting the array:
`signal1 = np.complex64(signal)`
`signal1.to_file("symbols.bin")`

I then phase-shift the signal by pi/2 (making the signal purely complex):
`signal2 = np.complex64(sig * np.exp(1j*np.pi/2)`
`signal2.to_file("symbols.bin")`

In gnuradio, I am running what is effectively an autocorrelation using the `Correlation Estimator` block. 
[image]

In the symbols field, I use `np.fromfile('symbols.bin',dtype=np.complex64)` to read the file.

The two signals should generate the same magnitude correlation at the output.

Here is the signal autocorrelation using a purely real vector:
[image]

Here is the autocorrelation using a purely complex vector:
[image]

I'm unable to find in the C++ code where the file is read into memory, but I believe that may be where the complex data is being dropped. I've tried using numpy.complex128 datatype as well, with the same result. I ran into this issue when attempting to correlate with an 8PSK preamble code, the block works fine for the common real BPSK preamble representations.

Using GR 3.8.2 on Fedora.

bistromath, 2021-06-11:
As you note, the file is being read into memory in np.fromfile(), not anywhere in the corr_est_cc code. What happens if, instead of using separate files for the real and rotated data, you initialize the correlator with `np.fromfile('symbols.bin') * np.exp(1j*np.pi/2)`? You can use a variable block to store that vector for use in both a vector source and the correlator. It would be an interesting experiment for you to make the phase rotation a variable and play with changing it during runtime.

sodaBitters, 2021-06-12:
Hi Nick, 
Thanks for the idea, I'm unable to change/multiply the data during runtime - perhaps because the `np.fromfile()` only gets evaluated once at initialization?

I was actually able to get it to work by re-casting to `np.complex128` within the block. If I recast to `np.complex64`, it doesn't work, so it seems that complex doubles are required for this block to work (does that make sense?) 
i.e. in the Symbols field I explicitly re-cast: 
`np.complex128(np.fromfile('symbols.bin',dtype=np.complex64))`

dkozel, 2021-09-29:
I'm not sure why the block wasn't accepting symbols in `np.complex64` format. The underlying implementation is using C++ `std::complex<float>` which should be interoperable with `np.complex64`.
https://www.gnuradio.org/doc/doxygen/classgr_1_1digital_1_1corr__est__cc.html

At minimum it sounds like this is another case where increasing the data type documentation and asserts would be helpful.
https://wiki.gnuradio.org/index.php/Correlation_Estimator

marcusmueller, 2024-01-28:
Can't reproduce, but that might be because I'm now using GR 3.10/main, which uses a different method of wrapping numpy objects (pybind11). So, closing for now. If the problem persists, please reopen

closed 2024-01-28 by marcusmueller