gnuradio/gnuradio #4761: OOT module import: Inadequate error handling

errikosmes, 2021-06-12:
I recently had a problem where a mistake I did in one c++ block that I developed caused every other c++ block in my module not to be imported without any warning (I just got a python error from my main script saying that the submodule I tried to import didn't exist). 

The mistake that I did wasn't strictly an error in c++ terms (I declared a forecast function without defining it), so it didn't get caught by the compiler. However it made the swig generated python submodule to fail completely. I managed to find the issue with the help of the community over at Element (thank you very much guys!), but the error was really hard to find. Installing a module that makes every other disappear was a pretty weird issue, at least for me.

Having had a think about why I never saw any import error messages that could have helped find the issue I stumbled upon this try-except bit on the `__init__.py` file generated by gr_modtool for my module:

```
# import swig generated symbols into the mymodule namespace
try:
    # this might fail if the module is python-only
    from .mymodule_swig import *
except ImportError:
    pass  
```

In my opinion that's kind of a dangerous use of a try-expept statement because in my case it hid away the error that led my to the solution (I had to import my module manually in a python shell to see it). I think it would be better to at least have it like that: 
```
# import swig generated symbols into the mymodule namespace
try:
    # this might fail if the module is python-only
    from .mymodule_swig import *
except ImportError as error:
    print(error)  
```
I get that this will print an error even when there's not a problem if the user has only defined python blocks, so we do need to find a better solution from what I'm proposing. But in the end the first version is just not right for me.

velichkov, 2021-06-14:
Hi @errikosmes,

Welcome to gnuradio and many thanks for the detailed report! In the `master` and `maint-3.9` branches this problem is solved by changing `ImportError` to `ModuleNotFoundError`. 

@willcode, I think that #3926 needs to be cherry-picked in `maint-3.8`.

willcode, 2021-06-14:
ModuleNotFoundError is not in Python2, so we would need a different change. Suggestions?

errikosmes, 2021-06-14:
Hi @velichkov,

Good to hear that the issue is getting handled in the newer versions. I'm still on 3.8.3.0 because I'm using a LimeSDR and their blocks are still 3.7 or 3.8 only.

I tested it and I can confirm that `ModuleNotFoundError` would had raised the error that led me to the solution.

For Python2 @willcode I have no idea unfortunately.

velichkov, 2021-06-15:
> ModuleNotFoundError is not in Python2, so we would need a different change.

OK, I see.

> Suggestions?

One possible alternative soltuion is to link the OOT's shared object with ` -Wl,--no-undefined` flag on the platforms that supports this (Linux with gcc and clang) or `-Wl,-undefined,error` on OSX. I don't know about Windows and various BSDs.

In `gr-utils/python/modtool/templates/gr-newmod/lib/CMakeLists.txt` something like this needs to be added.
```cmake
if (LINUX)
  string(APPEND CMAKE_SHARED_LINKER_FLAGS " -Wl,--no-undefined")
elseif(APPLE)
  string(APPEND CMAKE_SHARED_LINKER_FLAGS " -Wl,-undefined,error")
endif()
```
This way you will get an error when building the OOT.

willcode, 2021-07-13:
Maybe I'll backport the original PR so it uses ModuleNotFoundError for Python 3 only and keeps the current behavior for Python 2.

marcusmueller, 2024-01-28:
Solved in 3.9

closed 2024-01-28 by marcusmueller