gnuradio/gnuradio #4765: Convert hier_block2_detail to logging

marcusmueller, 2021-06-12:
largest occurrence of cout is in hier_block2_detail, for no good reason!

would be relatively easy to replace with `GR_LOG_INFO(d_logger,…)`, but it's mechanic.

Partially addresses #4763

ronalddadds, 2021-06-18:
Hi @marcusmueller, I've reviewed the work that needs to be completed and I'm happy to take this work on. Can you assign this issue to me?

I'll use PR #4764 as an example to work from.

dkozel, 2021-06-19:
Hi @ronalddadds. Thanks! If you have any questions here or https://chat.gnuradio.org are the best places to ask.

ronalddadds, 2021-06-20:
Thank you @dkozel for assigning this to me, I'll be sure to ask questions either here or on the GNU Radio chat link you provided.

ronalddadds, 2021-06-23:
I encountered a registration error on chat.gnuradio.org (an error on sending a validation email - and I double-checked my email!), but I will still communicate here.

My current method is to replace std::cout messages with GR_LOG_INFO as below:
``` diff
if (HIER_BLOCK2_DETAIL_DEBUG)
-    std::cout << "connect: dst is hierarchical, setting parent to " << this
-              << std::endl;
+    GR_LOG_INFO(d_logger,
+        boost::str(boost::format("connect: dst is hierarchical, setting parent to %s") % this));
```

Is this in line with expectations for this issue?

marcusmueller, 2021-06-25:
it really is! Thank you!

ronalddadds, 2021-06-26:
You're welcome! There was a _make test_ issue in my pull request #4810, so I'll fix that and let you know when it's done!

SWITCHin2, 2024-03-12:
@marcusmueller Hi there, i have tried to accomplish the 3rd sub-task mentioned in this issue and raised the PR for the same #7186 kindly look into it

SWITCHin2, 2024-03-12:
@marcusmueller I have raised a new PR with all the changes that you have asked to do , PR : #7188

StudHamza, 2024-03-27:
I think we can close this issue, PR #7188 fixed the last bullet point

closed 2024-03-27 by willcode