gnuradio/gnuradio #4770: pmt.mp(True).is_bool() is False

marcusmueller, 2021-06-15:
This seems to be a bug.

jacobagilbert, 2021-06-15:
I can confirm this on master and maint-3.9. 3.8 does not have is_bool().

marcusmueller, 2021-06-16:
I just found something confusing:

`block.cc`:
```c++
void block::system_handler(pmt::pmt_t msg)
{
…
        d_finished = pmt::to_long(pmt::cdr(msg));
…}
```

notice the `to_long` in there: that fails on PMTs that `is_bool()`

in `notify_msg_neighbors`, however:

```c++
blk->post(d_system_port, pmt::cons(d_pmt_done, pmt::mp(true)));
```

Is it possible we currently can't shut down message-based flow graphs or am I misreading this?

willcode, 2021-07-25:
@jacobagilbert Why didn't adding a `mp()` for bool work? Do you think we should make it work?

jacobagilbert, 2021-07-26:
@willcode 

I never ran this to ground, but the following patch does not work https://pastebin.com/TXbnsRDp

```python
>>> import pmt
>>> pmt.mp(True)
1
```

This calls one of the integer PMT overloads in pmt_sugar.h for some reason....seems like something pybind related. I see no reason *not* to add it to MP (if possible).

marcusmueller, 2024-01-28:
bug remains persistent

ffu, 2026-11-14:
This issue can be fixed by:
1. Adding another overloaded function mp(bool x) in pmt_sugar.h
2. Adding the binding for num 1 in pmt_sugar_python.cc. The crucial trick is that the binding for the new bool parameter must be listed BEFORE the ones for integer parameters. It's a quirk of pybind and how it attempts overloaded functions in the order of binding registration, and how it will accept boolean args as integers. @jacobagilbert added the binding at the very end, after the integer versions, which is why it didn't work. I'll also test if using py::arg.noconvert() helps.

Unfortunately, this bug fix would cause additional problems because the code has assumed this bug for a while. This includes QA code and the block definition mentioned in #4775. I'll keep looking into this, as I'm getting some freezes in QA code. Also, I'm not sure how to handle the pydoc template header.