gnuradio/gnuradio #4773: Unable to coerce endpoints when making a block in OOT module

armin485, 2021-06-15:
Hi Guys.
I've just installed GNURadio 3.9.2.0 and its installed well and working fine up to now.
But when I tried to make a block in an OOT module, I encountered this error and honestly I couldnt find any usefull solution in the internet. Here is the console output when trying to run the flowgraph:

Generating: '/home/aghani/Documents/gr-myoot/examples/myblock_test.py'

Executing: /usr/bin/python3 -u /home/aghani/Documents/gr-myoot/examples/myblock_test.py

Traceback (most recent call last):
  File "/usr/local/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 31, in wrapped
    endp = [(p.to_basic_block(), 0) if hasattr(p, 'to_basic_block')
  File "/usr/local/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 32, in <listcomp>
    else (p[0].to_basic_block(), p[1]) for p in points]
AttributeError: 'myoot.myoot_python.myblock' object has no attribute 'to_basic_block'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/aghani/Documents/gr-myoot/examples/myblock_test.py", line 196, in <module>
    main()
  File "/home/aghani/Documents/gr-myoot/examples/myblock_test.py", line 174, in main
    tb = top_block_cls()
  File "/home/aghani/Documents/gr-myoot/examples/myblock_test.py", line 143, in __init__
    self.connect((self.analog_const_source_x_0, 0), (self.myoot_myblock_0, 0))
  File "/usr/local/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 34, in wrapped
    raise ValueError("Unable to coerce endpoints: " + str(err))
ValueError: Unable to coerce endpoints: 'myoot.myoot_python.myblock' object has no attribute 'to_basic_block'

>>> Done (return code 1)

And let me add this that my block is the simplest block I could implement. If you think that it helps to find the bug let me know.
Many Thanks

jalgroy, 2021-07-06:
Any solution or workaround to this? I'm having the same issue with GNU Radio 3.9.2.0 and a newly created sync block in C++.

wsx000, 2021-07-06:
hi ,i‘ve met the same problem with you in recent days. and i finally found the solution today, but i met a new bug after this, following is my experience：
i find the solution at a official doc: [https://wiki.gnuradio.org/index.php/GNU_Radio_3.9_OOT_Module_Porting_Guide](url)
at the end of this doc, it says：

`TypeError: 'modulename_python.blockname' object is not subscriptable
This is caused by an incomplete inheritance chain specified in the binding declaration of the block.

Instead of

   py::class_<blockname,
              std::shared_ptr<blockname>>(m, "blockname", D(blockname))
Try something like (taking into account your block type)

   py::class_<blockname,
              gr::sync_block,
              gr::block,
              gr::basic_block,
              std::shared_ptr<blockname>>(m, "blockname", D(blockname))`

after doing that , this problem is eventually solved.
but when i rerun the commend `make test`, a new bug happend,and i can't fix this bug now, the main problems is :
ImportError: generic_type: type "square_ff" referenced unknown base type "gr::block"
following is the whole log after i run commend `ctest -V -R square`:

UpdateCTestConfiguration  from :/home/wsx/temp/gr-mymodule/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/wsx/temp/gr-mymodule/build/DartConfiguration.tcl
Test project /home/wsx/temp/gr-mymodule/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: qa_square_ff

1: Test command: /usr/bin/sh "/home/wsx/temp/gr-mymodule/build/python/qa_square_ff_test.sh"
1: Test timeout computed to be: 10000000
1: Traceback (most recent call last):
1:   File "/home/wsx/temp/gr-mymodule/python/qa_square_ff.py", line 12, in <module>
1:     from mymodule import square_ff
1:   File "/usr/local/lib/python3/dist-packages/mymodule/__init__.py", line 18, in <module>
1:     from .mymodule_python import *
1: ImportError: generic_type: type "square_ff" referenced unknown base type "gr::block"
1: 
1: During handling of the above exception, another exception occurred:
1: 
1: Traceback (most recent call last):
1:   File "/home/wsx/temp/gr-mymodule/python/qa_square_ff.py", line 18, in <module>
1:     from mymodule import square_ff
1:   File "/usr/local/lib/python3/dist-packages/mymodule/__init__.py", line 18, in <module>
1:     from .mymodule_python import *
1: ImportError: generic_type: type "square_ff" referenced unknown base type "gr::block"
1/1 Test #1: qa_square_ff .....................***Failed    0.19 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.19 sec

The following tests FAILED:
	  1 - qa_square_ff (Failed)
Errors while running CTest

Does anyone know how to solve this problem？

mormj, 2021-07-08:
I think @wsx000 has the solution for the original issue here.  As for the secondary issue, if GR was installed previously from the PPA, it could be related to:

https://www.mail-archive.com/discuss-gnuradio@gnu.org/msg73247.html

armin485, 2021-11-10:
After a while I installed GNURadio v3.9.4.0 and this issue is no longer available. I'll close this issue.

closed 2021-11-10 by armin485