gnuradio/gnuradio #4775: `block::system_handler` expects PMT long, gets PMT bool

marcusmueller, 2021-06-16:
Investigation #4770  I found something confusing:

`block.cc`:
```c++
void block::system_handler(pmt::pmt_t msg)
{
…
        d_finished = pmt::to_long(pmt::cdr(msg));
…}
```

notice the `to_long` in there: that fails on PMTs that `is_bool()`

in `notify_msg_neighbors`, however:

```c++
blk->post(d_system_port, pmt::cons(d_pmt_done, pmt::mp(true)));
```

Is it possible we currently can't shut down message-based flow graphs or am I misreading this?

jacobagilbert, 2021-06-16:
Based on my looking into #4770, `pmt::mp(true)` returns an integer type PMT object, so this will need to get updated accordingly eventually.

Sometimes two bugs make functional code.

marcusmueller, 2024-01-28:
yep, but as long as #4770 is still present and this does *not* break, we still have a bug.