gnuradio/gnuradio #4781: Flowgraph doesn't update properly if samp_rate if controlled by another variable

drmpeg, 2021-06-17:
If the samp_rate variable is a function of another variable, the flow graph is not updated properly when the variable is changed (and OK is clicked). Here's the example flow graph. samp_rate is set to (symbol_rate * 2). When symbol_rate is changed (in this case from 32k to 48k), samp_rate in all the blocks is not updated.

I've bisected this behavior to commit https://github.com/gnuradio/gnuradio/commit/c4f46bfa6d790c57018f9c6fca264bb9e01a2854

[image]

solomonbstoner, 2021-06-18:
I can replicate your described behavior when I press the `Ok` button immediately after changing `symbol_rate`. 
When I click `Apply` then `Ok` after changing `symbol_rate`, the Sample Rate in all the blocks get updated. Is it the same behavior on your side?

drmpeg, 2021-06-18:
Yes, that's correct. Only happens when just ```OK``` is clicked. But that's how I normally use the dialog.

solomonbstoner, 2021-06-18:
Wasnt suggesting you change how you use the dialog haha. Im trying to narrow down the behavior.
So it appears that `Apply` correctly applies the change in Symbol Rate to all the Sample Rate parameters in the blocks regardless of whether I press `Ok` or `Cancel` afterwards. So we can narrow the search to the difference between the `Ok` behavior and `Apply` behavior. Ill take a look at this issue too with you

PR #4784 (2021-06-19): grc: Update flowgraph vars controlled by vars
Fixes #4781. When a variable is controlled by another variable,
updating the latter does not update the former in the flowgraph when
`Ok` is pressed. Caused by commit c4f46bf. This commit patches the bug.
commit bfb5e3ba (2021-06-19): grc: Update flowgraph vars controlled by vars

Fixes #4781. When a variable is controlled by another variable,
updating the latter does not update the former in the flowgraph when
`Ok` is pressed. Caused by commit c4f46bf. This commit patches the bug.
PR #4805 (2021-06-21): grc: Update flowgraph vars controlled by vars (backport to maint-3.9)
Fixes #4781. When a variable is controlled by another variable,
updating the latter does not update the former in the flowgraph when
`Ok` is pressed. Caused by commit c4f46bf. This commit patches the bug.

(cherry picked from commit bfb5e3baa88763adb9e40ae3c511e317695fedb5)

Backport https://github.com/gnuradio/gnuradio/pull/4784
commit 77195790 (2021-06-21): grc: Update flowgraph vars controlled by vars

Fixes #4781. When a variable is controlled by another variable,
updating the latter does not update the former in the flowgraph when
`Ok` is pressed. Caused by commit c4f46bf. This commit patches the bug.

(cherry picked from commit bfb5e3baa88763adb9e40ae3c511e317695fedb5)
closed 2021-06-19 by mormj