gnuradio/gnuradio #4782: gr-zmq blocks are not compatible with zmq devices

amymalady, 2021-06-17:
## Overview

The [base_sink_impl](https://github.com/gnuradio/gnuradio/blob/master/gr-zeromq/lib/base_impl.cc#L83) enforces that the sink blocks bind their ports. ZMQ does *not* enforce any conventions that certain zmq socket types bind the port.

Similarly, [base_source_impl](https://github.com/gnuradio/gnuradio/blob/master/gr-zeromq/lib/base_impl.cc#L157) enforces that the source blocks connect their ports. ZMQ does *not* require that certain zmq socket types connect to the port. 

## Issues

[ZMQ devices](https://learning-0mq-with-pyzmq.readthedocs.io/en/latest/pyzmq/devices/devices.html) are a useful construct for managing ZMQ networks; the current assumptions on sink/source bind/connect are not compatible with ZMQ devices. 

As an example, consider [ZMQ Forwarders](https://learning-0mq-with-pyzmq.readthedocs.io/en/latest/pyzmq/devices/forwarder.html). These devices act like a pub-sub proxy server that allows multiple publishers to route to multiple subscribers through a stable hub. 

[image]

In this model, typically the forwarder_device **binds** both the front and backend ports (since it is acting as a server), see code snippet below from [zmq documentation](https://learning-0mq-with-pyzmq.readthedocs.io/en/latest/pyzmq/devices/forwarder.html)

[image]

When trying to integrate a GnuRadio flowgraph with a larger framework, our ZMQ PUB ports are not compatible with this standard ZMQ device. 

There are analogous devices  for [XREP/XREQ](https://learning-0mq-with-pyzmq.readthedocs.io/en/latest/pyzmq/devices/queue.html) and  [PUSH/PULL](https://learning-0mq-with-pyzmq.readthedocs.io/en/latest/pyzmq/devices/streamer.html) models that are also not compatible. 

## Proposed Solution

Update the base_source_impl and base_sink_impl to **default** to bind/connect (respectively), but allow for configuration that supports either binding or connecting the socket.

willcode, 2021-06-17:
This makes sense to add. I think we already allow bind/connect selection elsewhere.

duggabe, 2021-06-18:
Within the GR world, ZMQ sinks **always** bind and ZMQ sources **always** connect.
The PUSH/PULL models are consistent within GR.
I think losing that consistency would be detrimental.
The GR implementation of ZMQ is not always the same as ZeroMQ in the general sense.
IMO, the external Python program should conform to the conventions of GR. See https://wiki.gnuradio.org/index.php/Understanding_ZMQ_Blocks

amymalady, 2021-06-18:
If you'll note the ZMQ Forwarder device you'll notice three publisher ports that publish to a single subscriber. The subscriber *must* bind the port as each publisher is sharing the same port. If all three publishers were flow graphs (as is my case), there is no way for the ZMQ Forwarder to adhere to the GR convention as the GR convention enforces 1:N publisher: subscribers and cannot support N:1 publishers: subscriber.

Hope that clarifies the issue further. 

Thanks

skoold2003, 2022-04-10:
@amymalady , I've just run into the same issue where I need to connect using the PUB sink and bind on the SUB source. I was going to post a new feature request but found this first. I wholeheartedly agree with you that adding this feature is needed. Did you find a workaround for this that doesn't involve too much coding effort?

amymalady, 2022-04-10:
Hey!

We wrote a local OOT gr-zmq replacement that implements defaults that preserve gr convention but allow for overwriting that behavior.

I have an outstanding action to put in a PR for these changes, but haven't done it yet! 

Fwiw the changes took <4 hours starting with the gr-zmq lib and adding and optional bool for bind/connect that defaults appropriately.

skoold2003, 2022-04-10:
OK good.. since you're putting in a PR I'm just going to make a couple of python blocks. Thanks

willcode, 2023-02-27:
@amymalady Did you ever get to work on this PR? I've found a user for client/server selection myself.

willcode, 2023-03-22:
@amymalady Take a look at #6596.

PR #6596 (2023-03-24): zmq: user-selectable connect/bind for all zmq blocks
## Description
Message-oriented ZMQ blocks already have selectable connect/bind. Here, the stream-oriented blocks are given the same option. Some refactoring of the base classes was necessary. The API and default behavior is unchanged.

## Related Issue
Various ZMQ improvements were dicussed in #6559. 
Specifically fixes #4782.

## Which blocks/areas does this affect?
ZMQ stream-oriented blocks.

## Testing Done
PUB (connect) to SUB (bind) tested in a flowgraph.

QA tests and documentation TBD.

## Checklist
- [x] I have read the [CONTRIBUTI [... cut]
closed 2023-03-24 by willcode