gnuradio/gnuradio #4783: [maint-3.9] relocation R_X86_64_PC32 against symbol `PyFloat_Type' can not be used when making a shared object; recompile with -fPIC

RobertSwirsky, 2021-06-17:
I'm trying to build maint-3.9 on a Ubuntu system; I've done this at least half a dozen times in the past

    commit f2af47f58f223ee0f935463d9b1fe6482c168e81 (HEAD -> maint-3.9, origin/maint-3.9)

On clean install of a Ubuntu 20.04.2 LTS,all the prerequisites were installed per the directions here:

    https://wiki.gnuradio.org/index.php/InstallingGR#For_GNU_Radio_3.9_and_Master_Branch

The build fails:

```
/usr/bin/ld: /usr/local/lib/libpython3.8.a(floatobject.o): relocation R_X86_64_PC32 against symbol `PyFloat_Type' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/build.make:970: gnuradio-runtime/lib/libgnuradio-runtime.so.v3.9.2.0-5-gf2af47f5] Error 1
make[1]: *** [CMakeFiles/Makefile2:4290: gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/all] Error 2
```

I'm trying to force the -fPIC flag in to see what happens, but I've never had the instructions on how to do a build not "just work".

I need to build from sources because I'm developing some custom blocks.

velichkov, 2021-06-17:
Hi @RobertSwirsky,

Welcome to Gnuradio! 

> /usr/bin/ld: /usr/local/lib/libpython3.8.a(floatobject.o):

It looks like you have a custom `python3.8` installed in `/usr/local/lib`, the static library (`libpython3.8.a`) is available but most probably the shared object (`libpython3.8.so`) is missing. Debina/Ubuntu packages does not install files in `/usr/local/` so most probably this python has been installed from source. 

Do you need `gnuradio` to be built with this particular `python` and if yes can you provide some more information on how did you install it? 

If you don't then you can probably remove it, the remove the gnuradio build directory and stat from scratch. Alternatively you need to fight with `cmake` to use the python in `/usr/lib/`

RobertSwirsky, 2021-06-17:
Thank you!  I have no particular need for this particular version of Python -- it was just what was in my standard machine provisioning configuration. I'll remove these /usr/local Pythons and try again.

I like to build things like "GnuRadio" on the happy path, without hacking the cmake files, etc.

RobertSwirsky, 2021-06-17:
Ok! I removed that private python build, did another cmake and make, and it all works! Thanks for the tip.

closed 2021-06-17 by RobertSwirsky

imkzh, 2022-07-23:
For those who want to compile grc with custom built python, make sure you compiled your python with `--enable-shared`, if you have already installed the custom python without `--enable-shared`, you should manually delete the file named `libpythonX.Y.a` under your prefix (default to /usr/lib/local/libpythonX.Y.a), and redo the `cmake` then `make`. 

(Is this a bug in cmake script making decision to link `.a` file when both `libpythonX.Y.a` and `libpython.X.Y.so` are presented?)