gnuradio/gnuradio #4786: Feature request: automated tests on MacOS and Windows

NeuroForLunch, 2021-06-19:
I've seen some services that allow automated testing on MacOS and Windows so it might be a good idea to test on those platforms as well since they are more popular than Ubuntu, Fedora, CentOS, and Debian combined.

dkozel, 2021-06-19:
Hi @NeuroForLunch This is in progress!
https://github.com/gnuradio/gnuradio/pull/4410

Our difficulty so far has been that there's more potential users on macOS and Windows, but far far fewer people on those platforms testing and contributing to GNU Radio. This has become a painful circle where GNU Radio on those OSs is unstable or becomes out of date and so potential contributors stop using it on those OSs, so the project gets little testing and few contributions. Happily it seems to be improving quickly over the last two years.

NeuroForLunch, 2021-06-19:
Interesting to hear about how the end users are on macOS and Windows while the producers are on the linux platforms but it makes sense. Multiple people have recently been updating the homebrew version to install GNUradio 3.9 and there's almost 900 installs over the past 30 days and over 11,000 installs over the past year so there must be quite a bit of users on OSX. I think it might be good to have someone who makes sure all the distribution points on all platforms are up to date because that makes it much easier to install GNURadio versus installing it from source. Even on Ubuntu with apt I ran into problems: UHD is a dependency of some OOT modules for 3.8 yet UHD has a dependency of an old version of GNURadio 3.8 and Python 2 which would mess up the installed version of the latest maint-3.8 for me. I'm trying to get soapysdr and soapyrtlsdr on brew because the 3.9 version that installs via brew doesn't contain soapy as it's missing those dependencies and most people probably have the $30 dongles which makes the brew installation useless for them as they can't use it with their hardware.

closed 2021-06-19 by NeuroForLunch