gnuradio/gnuradio #4788: Flowgraph doesn't update disabled blocks properly if samp_rate is controlled by another variable

solomonbstoner, 2021-06-19:
Bug found by @drmpeg. Related to Issue #4781, except this is specific to disabled blocks. 

Download [this flowgraph](http://www.w6rz.net/samptest.grc). There are two variables, `symbol_rate` and `samp_rate`. `samp_rate = symbol_rate * 2`. 

#### Expected behavior
When the `symbol_rate` is changed, and `Apply` is pressed, the flowgraph should reflect the new `samp_rate` for all blocks.

#### Actual behavior
The following behaviors were noted:
1. Pressing just the `OK` button doesn't update the disabled block. Either that or there's a lag in the updating. For example, setting symbol_rate to 40e3 at first and pressing Ok wouldnt update the disabled block to 80e3. But after I re-set symbol_rate to 50e3 and press Ok, the disabled block gets set to 80e3.
2. `Apply` updates the disabled block, but only after `OK` is clicked. 
3. Pressing `Apply` + `Cancel` doesnt update the disabled block for some reason (while the enabled blocks are updated). 

Everyone is welcome to try out that flowgraph and report any other unexpected behaviors. It helps us narrow down the cause :)

dl1ksv, 2021-06-30:
I can reproduce this behaviour.
But what is the expected behaviour?
Should disabled blocks be updated if enabled blocks are updated or should the be untouched.

If I open this flowgraph in grc the disabled throttle displays:
Sample Rate: samp_rate.
That means the the value is not evaluated.  But open and close a block forces the disabled block to evaluate it's value.

My proposal: Do not evaluate disabled blocks until they will be enabled

solomonbstoner, 2021-07-15:
Nice! Thanks for resolving the issue :)

PR #4851 (2021-07-19): grc: update disabled blocks if they depend on others
If we want disabled blocks to be updated properly,
we have to consider all blocks when updating the fg not only the enabled.

This does the patch of Flowgraph.py and fixes #4788.

But a problem remains.
If you use the example in #4788 and disable the symbol_rate block,
only the samp_rate block 'turns to red'.
The other blocks don't change.
Only if you open and close arbitrary block, the other blocks are updated.

This is fixed by the patch of Application.py.
commit fbab6c38 (2021-07-19): grc: update disabled blocks if they depend on others

If we want disabled blocks to be updated properly,
we have to consider all blocks when updating the fg not only the enabled.

This does the patch of Flowgraph.py and fixes #4788.

But a problem remains.
If you use the example in #4788 and disable the symbol_rate block,
only the samp_rate block 'turns to red'.
The other blocks don't change.
Only if you open and close an arbitrary block, the other blocks are updated.

This is fixed by clearing the namespace in renew_namespace in FlowGraph.py.

grc: update disabled blocks
PR #4923 (2021-07-19): grc: update disabled blocks if they depend on others (backport to maint-3.9)
If we want disabled blocks to be updated properly,
we have to consider all blocks when updating the fg not only the enabled.

This does the patch of Flowgraph.py and fixes #4788.

But a problem remains.
If you use the example in #4788 and disable the symbol_rate block,
only the samp_rate block 'turns to red'.
The other blocks don't change.
Only if you open and close an arbitrary block, the other blocks are updated.

This is fixed by clearing the namespace in renew_namespace in FlowGraph.py.

grc: update disabled block [... cut]
commit 68d7189a (2021-07-19): grc: update disabled blocks if they depend on others

If we want disabled blocks to be updated properly,
we have to consider all blocks when updating the fg not only the enabled.

This does the patch of Flowgraph.py and fixes #4788.

But a problem remains.
If you use the example in #4788 and disable the symbol_rate block,
only the samp_rate block 'turns to red'.
The other blocks don't change.
Only if you open and close an arbitrary block, the other blocks are updated.

This is fixed by clearing the namespace in renew_namespace in FlowGraph.py.

grc: update disabled blocks

(cherry picked from  [... cut]
closed 2021-07-19 by mormj