gnuradio/gnuradio #4811: LFTX/LFRX daughterboard : invalid daughterboard frontend name: A

wsx000, 2021-06-24:
hi,i have a problem when i use X310 with two daughterboards named LFTX and LFRX respectively。i set the gnuradio parameter “Mb0： subdev  spec” with "B:A"(both LFTX and LFRX daughterboard  are installed at slot B on motherboard X310)，but it raised error “RuntimeError: RuntimeError: Invalid daughterboard frontend name: A” when i run the flowgraph(so did “B:B”)，and this error appered on both LFTX and LFRX daughterboard。but when i set “Mb0： subdev  spec” with "B:BA" or “B:AB”， it worked well，i dont know what‘s wrong with my equipment。

**the complete error information is :**
Traceback (most recent call last):
  File "/home/wsx/Desktop/FM_Radio/FM_receiver.py", line 299, in <module>
    main()
  File "/home/wsx/Desktop/FM_Radio/FM_receiver.py", line 275, in main
    tb = top_block_cls()
  File "/home/wsx/Desktop/FM_Radio/FM_receiver.py", line 115, in __init__
    self.uhd_usrp_source_0.set_subdev_spec('B:A', 0)
  File "/usr/local/lib/python3/dist-packages/gnuradio/uhd/uhd_swig.py", line 3575, in set_subdev_spec
    return _uhd_swig.usrp_source_sptr_set_subdev_spec(self, spec, mboard)
RuntimeError: RuntimeError: Invalid daughterboard frontend name: A
QFileSystemWatcher::removePaths: list is empty
QFileSystemWatcher::removePaths: list is empty

**following is the log when i executed commend "uhd_usrp_probe" in terminal:**
[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_3.15.0.HEAD-0-gaea0e2de
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 200 MHz
[INFO] [GPS] No GPSDO found
[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1305 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1298 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)
  _____________________________________________________
 /
|       Device: X-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: X310
|   |   revision: 11
|   |   revision_compat: 7
|   |   product: 30818
|   |   mac-addr0: 00:80:2f:32:2a:42
|   |   mac-addr1: 00:80:2f:32:2a:43
|   |   gateway: 192.168.10.1
|   |   ip-addr0: 192.168.10.2
|   |   subnet0: 255.255.255.0
|   |   ip-addr1: 192.168.20.2
|   |   subnet1: 255.255.255.0
|   |   ip-addr2: 192.168.30.2
|   |   subnet2: 255.255.255.0
|   |   ip-addr3: 192.168.40.2
|   |   subnet3: 255.255.255.0
|   |   serial: 320AFFF
|   |   FW Version: 6.0
|   |   FPGA Version: 36.0
|   |   FPGA git hash: fde2a94
|   |   RFNoC capable: Yes
|   |   
|   |   Time sources:  internal, external, gpsdo
|   |   Clock sources: internal, external, gpsdo
|   |   Sensors: ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: 0
|   |   |   |   Name: Unknown (0xffff) - 0
|   |   |   |   Antennas: 
|   |   |   |   Sensors: 
|   |   |   |   Freq range: 0.000 to 0.000 MHz
[... 122 lines cut ...]
|   |   |   |       TX Frontend: B
|   |   |   |   Name: LFTX (B)
|   |   |   |   Antennas: 
|   |   |   |   Sensors: 
|   |   |   |   Freq range: -32.000 to 32.000 MHz
|   |   |   |   Gain Elements: None
|   |   |   |   Bandwidth range: 32000000.0 to 32000000.0 step 0.0 Hz
|   |   |   |   Connection Type: Q
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: B
|   |   |   |   Name: ad9146
|   |   |   |   Gain Elements: None
|   |     _____________________________________________________
|   |    /
|   |   |       RFNoC blocks on this device:
|   |   |   
|   |   |   * DmaFIFO_0
|   |   |   * Radio_0
|   |   |   * Radio_1
|   |   |   * DDC_0
|   |   |   * DDC_1
|   |   |   * DUC_0
|   |   |   * DUC_1

dkozel, 2021-06-26:
Hi @wsx000 I'm pretty sure this is a change in the behavior of the UHD driver. Try setting the antenna to "A" or "B" rather than using the subdev spec.
https://files.ettus.com/manual/page_dboards.html#dboards_basicrx

willcode, 2021-07-09:
The gr-uhd layer passes the subdev spec directly to UHD
```
void usrp_source_impl::set_subdev_spec(const std::string& spec, size_t mboard)
{
    return _dev->set_rx_subdev_spec(spec, mboard);
}
```
Please reopen this issue if you believe this is a GNU Radio (vs UHD driver) problem.

closed 2021-07-09 by willcode