gnuradio/gnuradio #4825: OOT QA tests fail unless module installed

antonott, 2021-06-29:
This issue has been discussed in a [thread on the mailing list](https://lists.gnu.org/archive/html/discuss-gnuradio/2021-06/msg00106.html).

To summarize, when creating an OOT module, the QA tests don't work correctly until we run `sudo make install`. This seems backwards because we should be able to run QA tests before we decide to install the module.

I have set up a [repo that illustrates the problem](https://github.com/antonott/oot_tests_failing). I originally discovered the issue when trying to work through the [OOT tutorial](https://wiki.gnuradio.org/index.php/OutOfTreeModules), but in the repo I instead use the module [gr-iqlevels](https://github.com/drmpeg/gr-iqlevels), to reduce the likelihood that the issue is just due to a mistake by me.

As Vasil points out in the mailing list, the issue seems to come from the migration from swig to pybind. See his analysis [here](https://lists.gnu.org/archive/html/discuss-gnuradio/2021-06/msg00119.html).

I am running Ubuntu 20.04.2 LTS, and installed GNU Radio 3.9.2.0 via PPA. I have the 2.5.0 version of pybind11-dev from the PPA.

dkozel, 2021-07-01:
Hi Anton,

Thanks for filing the issue. Fully agree that OOT tests should be able to run without installing. I think this was just checked or fixed in GNU Radio itself thanks to @balister testing it out. Very similar changes should be able to solve the issue in OOTs. Philip, do you remember what changes were needed?

bmarjanovic, 2021-07-19:
Hi,
Not sure if there were any changes on this. Due to vested interest I did some digging at what is missing. 

The problem is that module name is not found in PYTHONPATH. The generated python test harness tries to catch the exception and add the path to the locally built python bindings module. Seems like there are couple of things missing.
There is no file or directory with <module-name> that could be imported. And the directory containing compiled bindings (which I'd expect to be the module directory) does not contain __init__.pyX files. They are one level higher

In order to get the tests to work without installing the module I used a bit of a workaround - move a couple of files and add a symbolic link:
```
$ cd <module>/build/python
$ mv __init__.py* bindings/
$ ln -s <module> bindings

```
Running `make test` then works fine. I hope this can help someone fix the issue. Both my python and cmake knowledge are a bit short of what is probably needed to fix the build in reasonable time :)

BTW I'm running GR v3.9.0.0-163-g9b408129 locally built and installed to custom destination (not sure that makes difference, but  just in case)

PR #5279 (2021-11-17): modtool,cmake: make module dir for tests 
# Pull Request Details

## Description

Updated:

This change creates a module directory inside the build tree (`build/test_modules/modulename`). It consists of the build artifacts from `build/python/bindings` and `python/*`. `GrTest` then adds `build/test_modules` to the PYTHONPATH.

Old description:
This change makes CTest look for locally-built modules (i.e. during development, in the build directory) before installed ones.

Also adjust test module name in the template, though that's not ideal. I _think_ we could remove the `_python` suffix from the [... cut]
closed 2021-11-17 by mormj