gnuradio/gnuradio #4826: USRP N300 - Runtime error

nielsvanetten, 2021-06-29:
Dear all,

After a succesfull `uhd_usrp_probe` when trying to run the Ettus N300 on GNU radio I get the following error message:

```
Executing: /usr/bin/python3.6 -u /home/niels/top_block.py

[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-62-g7a3f1516
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=192.168.1.5,type=n3xx,product=n300,serial=320C34D,claimed=False,addr=192.168.10.2,master_clock_rate=125.0e6
Traceback (most recent call last):
  File "/home/niels/top_block.py", line 189, in <module>
    main()
  File "/home/niels/top_block.py", line 165, in main
    tb = top_block_cls()
  File "/home/niels/top_block.py", line 84, in __init__
    channels=list(range(0,1)),
  File "/home/niels/prefix-3.8/lib/python3/dist-packages/gnuradio/uhd/__init__.py", line 125, in constructor_interceptor
    return old_constructor(*args)
  File "/home/niels/prefix-3.8/lib/python3/dist-packages/gnuradio/uhd/uhd_swig.py", line 3298, in make
    return _uhd_swig.usrp_source_make(device_addr, stream_args, issue_stream_cmd_on_start)
RuntimeError: rpc::timeout: Timeout of 2000ms while calling RPC function 'get_num_xbars'

>>> Done (return code 1)
```
Note that I have no problem running the UHD receiver example program:
`/usr/local/lib/uhd/examples/rx_ascii_art_dft --args "master_clock_rate=125e6,mgmt_addr=192.168.1.151,addr=192.168.10.2" --freq 98.5e6 --rate 2.5e6 --gain 50 --ref-lvl="-50" --dyn-rng 90 --ant "RX2" --subdev "A:0"`

What I did notice in the arguments is that the UHD example runs:
`[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_4.0.0.0-185-g546ea894`
Instead of UHD_3.15.

Is GNU radio out of date? Or is something else going on?

nielsvanetten, 2021-07-02:
found the issue, N300 works only on version UHD_4.0.0.0-185-g546ea894.
Note that UHD 4.1 also gives a runtime error.

dkozel, 2021-07-03:
Hi @nielsvanetten I'm glad you found your issue. GNU Radio will use whatever version of UHD is installed already when GNU Radio is built. If you are installing GNU Radio from the PPA or Ubuntu default repositories then that version of UHD is fixed and cannot be changed. If you need a newer version of UHD to be used then GNU Radio needs to be built from source after installing the newer UHD.

It seems you have two different UHDs installed at the same time. The /usr/local/lib/uhd/ 4.0 version is probably one you built from source manually? and the /home/niels/prefix-3.8/lib one will be one built/installed by PyBOMBS, probably installed from the default Ubuntu repository.

Do you now have a working setup? If not the quickest way of untangling this is going to be to pick one method of installation (probably building UHD and GNU Radio from source) and uninstalling the others, then making sure that what is left is the correct 4.0+ UHD and whichever GNU Radio you prefer (3.9 is a good choice as it includes some of the RFNoC blocks by default).

The best place to get interactive help is going to be https://chat.gnuradio.org.

nielsvanetten, 2021-07-04:
Thanks Derek.

Yes this fixed the issue. I believe that there is some bug going on with the UHD installation on the N300, since it wont install UHD 4.1 on the N300 device. I have now fixed it by installing UHD 4.0 and reinstalling the latest version of GNU radio. Everything works. Thanks for the support.

closed 2021-07-04 by nielsvanetten

sumit429, 2022-08-23:
Executing: /usr/bin/python3 -u /home/signal/Desktop/gnuradio_sumit/usrp.py

[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=192.168.10.2,type=n3xx,product=n320,serial=3248C9D,claimed=False,addr=192.168.10.2,add=192.168.10.2
Traceback (most recent call last):
  File "/home/signal/Desktop/gnuradio_sumit/usrp.py", line 354, in <module>
    main()
  File "/home/signal/Desktop/gnuradio_sumit/usrp.py", line 332, in main
    tb = top_block_cls()
  File "/home/signal/Desktop/gnuradio_sumit/usrp.py", line 80, in __init__
    self.uhd_usrp_source_0 = uhd.usrp_source(
  File "/usr/lib/python3/dist-packages/gnuradio/uhd/__init__.py", line 125, in constructor_interceptor
    return old_constructor(*args)
  File "/usr/lib/python3/dist-packages/gnuradio/uhd/uhd_swig.py", line 2787, in make
    return _uhd_swig.usrp_source_make(device_addr, stream_args, issue_stream_cmd_on_start)
RuntimeError: rpc::timeout: Timeout of 2000ms while calling RPC function 'get_num_xbars'

I am also getting an same error. How to resolve this.

LawsonGu, 2023-01-06:
Change your &PYTHONPATH temporarily：
`export PYTHONPATH=/usr/local/lib/pythonXXX/site-packages/`
Like I'm using Python3.8, it could be like this:
`export PYTHONPATH=/usr/local/lib/python3.8/site-packages/`
It could work until you close the terminal

If you want this setting run up everytime you open terminal, adding the line below to the end of your $HOME/.bashrc file:
`export PYTHONPATH=/usr/local/lib/python3.8/site-packages/`
• If the PYTHONPATH environment variable is already defined with other folders in
your $HOME/.bashrc file, then add the line below to the end of your $HOME/.bashrc file to preserve the
current settings.
`export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.9/site-packages/`

marcivanmanalac, 2023-08-30:
I can confirm that this is still an issue, at least for the E310 series USRPs.
I fixed it by following @dkozel :)

Summary of what I did:
- Ensure UHD matches on both host and USRP
- Test the network connection by ssh and usd_usrp_probe --args addr=<IP_ADDR>
   NOTE: (network mode command is deprecated for UHD 4.4.0)
- Build and Install GNU Radio from source (because GNU Radio from default releases come with fixed UHD as stated)

I know this issue was old and closed, but still relevant and helpful.
Thanks!

hassanmaree, 2023-09-06:
dear marcivanmanalac, I have the ni usrp e313. I am also getting the same error. I installed uhd using pybomb and downloaded my flash card image using "sudo uhd_images_downloader -t sdimg -t e310 -t sg3". I can acces my device using ssh root@192.168.10.2. but the problem is when i run gnuradio-companion from root, it does not run and gives error command not found. when i try to run gnuradio from home it starts, but when i make a simple block diagram to receive signal. i get the same run time error discussed above.

marcivanmanalac, 2023-09-07:
> dear marcivanmanalac, I have the ni usrp e313. I am also getting the same error. I installed uhd using pybomb and downloaded my flash card image using "sudo uhd_images_downloader -t sdimg -t e310 -t [...]

Can you confirm that all of your UHDs(in host, gnuradio, and USRP) are the same verion?

When you execute a flow graph you should see the UHD version within the return of the terminal.

hassanmaree, 2023-09-07:
I used this link to install uhd and gnuradio.
 https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwiw0tuljJmBAxXkUkEAHftUCxQQFnoECBkQAQ&url=https%3A%2F%2Fkb.ettus.com%2FBuilding_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux&usg=AOvVaw262PyJzC2hCij8lCp1k53_&opi=89978449 

I had v4.4 uhd on host but I gues version on gnuradio was different which was 3.15. btw I have installed fresh ubantu system. I am doing it from start using following link https://wiki.gnuradio.org/index.php?title=Draft-AN-445

I have this image on my niusrp sdcard https://files.ettus.com/binaries/cache/e3xx/meta-ettus-v4.4.0.0/e3xx_e310_sg3_sdimg_default-v4.4.0.0.zip

marcivanmanalac, 2023-09-07:
> I used this link to install uhd and gnuradio. [...]

Yea, GNU Radio has UHD built in so you need to make GNURadio from source. I used GNURadio v3.11.0.0 which had UHD v4.4.0.0 on my ubuntu 20.04

hassanmaree, 2023-09-07:
https://wiki.gnuradio.org/index.php?title=LinuxInstall#Installing_Volk

I am using above link to install gnuradio? I am confused at this point. which version would this install? how can i make it gnu 3.11?

hassanmaree, 2023-09-07:
[image]

marcivanmanalac, 2023-09-08:
While in the cloned gnu radio repo, Run "git tag -l" to see all the versions. Then select a version to checkout using "git checkout <version_number>" then build it out from there

hassanmaree, 2023-09-08:
Following is the response when i try to run a simple receiption program. which show the gnu radio uhd is 4.4. the gnu radio version is 3.10.7. Moreover, the uhd host version is also 4.4. The image on usrp is also 4.4. the device can be ping also.

 Generating: '/home/grcusrp/gnuradio/build/first.py'

Executing: /usr/bin/python3 -u /home/grcusrp/gnuradio/build/first.py

Qt GUI: Could not restore geometry: restoreGeometry(self, Union[QByteArray, bytes, bytearray]): argument 1 has unexpected type 'NoneType'
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.4.0.HEAD-0-g5fac246b
[INFO] [MPMD FIND] Found MPM devices, but none are reachable for a UHD session. Specify find_all to find all devices.
Traceback (most recent call last):
  File "/home/grcusrp/gnuradio/build/first.py", line 160, in <module>
    main()
  File "/home/grcusrp/gnuradio/build/first.py", line 138, in main
    tb = top_block_cls()
  File "/home/grcusrp/gnuradio/build/first.py", line 71, in __init__
    self.uhd_usrp_source_0 = uhd.usrp_source(
RuntimeError: LookupError: KeyError: No devices found for ----->
Empty Device Address

hassanmaree, 2023-09-08:
the problem is when i run uhd find devices, initially i get following response. 

[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.4.0.HEAD-0-g5fac246b
--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    serial: 31370F4
    claimed: True
    fpga: n/a
    mgmt_addr: 192.168.10.2
    name: ni-e31x-31370F4
    product: e310_sg3
    reachable: No
    type: e3xx

hassanmaree, 2023-09-08:
when i run uhd usrp probe, i get following response. 

[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.4.0.HEAD-0-g5fac246b
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=192.168.10.2,type=e3xx,product=e310_sg3,serial=31370F4,name=ni-e31x-31370F4,fpga=n/a,claimed=False,addr=192.168.10.2
Error: rpc::timeout: Timeout of 10000ms while calling RPC function 'claim'

hassanmaree, 2023-09-08:
when i again run uhd find devices , i get no device found. 
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.4.0.HEAD-0-g5fac246b
No UHD Devices Found

willcode, 2023-09-08:
Please look for help on the USRP Users mailing list. This issue tracker is used for GNU Radio bugs and feature requests.

hassanmaree, 2023-09-09:
> Yea, GNU Radio has UHD built in so you need to make GNURadio from source. I used GNURadio v3.11.0.0 which had UHD v4.4.0.0 on my ubuntu 20.04

hi,could you please send me any link from where i should dowload and install my uhd and gnu radio? i have fpage image 4.4

hassanmaree, 2023-09-09:
uhd_find_devices
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.4.0.HEAD-0-g5fac246b
--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    serial: 31370F4
    claimed: True
    fpga: n/a
    mgmt_addr: 192.168.10.2
    name: ni-e31x-31370F4
    product: e310_sg3
    reachable: No
    type: e3xx

this is the response when i try to run uhd find devices