gnuradio/gnuradio #4827: problem in "QT GUI Frequency Sink" (freq_sink_c_impl.cc)

ali69550, 2021-06-30:
Hi,
In the code of "freq_sink_c_impl.cc" there is a problem:
instead of averaging abs(fft())^2 and then taking the logarithm,
it first calculates the logarithm of abs(fft())^2 and then averages!
as we all know log(a + b) != log(a) + log(b).
This causes a 2.5 dB lower result that can be easily shown by the following simple Matlab code:

M = 4; % M-PSK
L = 10000; % no. of averages
Nfft = 2000; % fft size
N = Nfft * L;
sbi_v = randi(M, N, 1) - 1; % integer symbols
sbc_v = pskmod(sbi_v, M, pi/4, 'gray'); % complex symbols
sbc_m = reshape(sbc_v, Nfft, N/Nfft); 
SBC_m = abs( fft(sbc_m, Nfft) ).^2;
SBC_log_m = 10*log10(SBC_m);
plot( [mean( SBC_log_m, 2) 10*log10( mean( SBC_m, 2) )] )
grid on

willcode, 2021-07-09:
I traced this one through, then closed the window before I save the comment. Anyway, the gist of it was that there is a scaling factor of 2.5 when you get down to the volk psd kernel called by fft. Could you check to see whether that is the missing 2.5dB you mention above? Seems like quite a coincidence otherwise.

ali69550, 2021-07-10:
I submit it to the mailing list and "Fabian Schwartau" answered me this:

"OK, yes, I think this is a bug.
In
https://github.com/gnuradio/gnuradio/blob/master/gr-qtgui/lib/freq_sink_c_impl.cc#L553
The data is converted using the fft() function and averaged after that.
But the fft() function already converts the transformed data into the
logarithmic scale in
https://github.com/gnuradio/gnuradio/blob/master/gr-qtgui/lib/freq_sink_c_impl.cc#L359
Averaging after the conversion to dB will yield wrong values, as
log(a+b) != log(a) + log(b).

Fabian"

FabianSchwartau, 2021-07-10:
> Anyway, the gist of it was that there is a scaling factor of 2.5 when you get down to the volk psd kernel called by fft.

I did not traced this down, but even if that is the case, this is wrong. There are probably scaling factors somewhere, to compensate the influence of the applied window (there is a button in the gui to turn this compensation on and off), but that does not make the result correct.
My answer on the mailing list explained the problem, I think. Want you want to do is to calculate the spectral power (density) values, average, and then convert them to dB.

willcode, 2021-07-10:
Here's how it's broken up in the implementation, for anyone who wants to look at it:
freq_sink_c_impl.cc/fft(): `psd(fftw_fft(data, norm=fft_size))`
freq_sink_c_impl.cc/work(): `exp_avg(fft)`
volk_32fc_s32f_power_spectrum_32f.h: psd = `10*log10(norm*mag2(data))`

The overall result is
`exp_avg(10*log10(fft_size*mag2(fft(data))))`

The factor of 2.5 I thought I was being used is not in the code path.

So,
- the average is being done after the log
- this is apparently mathematically incorrect
- averaging was implemented from a graphics point of view for smoothing, so this was probably not considered
- if the only result is a 2.5dB difference, the easiest solution would be to multiply somewhere, e.g., pass 2.5*fft_size to the psd function for normalization or pass in 2.5^2 as the rbw value (both hacks)
- the volk psd function does the mag2 and log in the same kernel, and the average is implemented in the GR code, so swapping the order is not a trivial change
- one solution would be to have a volk psd kernel that takes a running average and updates it
- the average is an exponential decay - question: does the math still hold?

willcode, 2021-07-10:
@marcusmueller would you be able to look at the math here?

FabianSchwartau, 2021-07-10:
>     * if the only result is a 2.5dB difference, the easiest solution would be to multiply somewhere, e.g., pass 2.5*fft_size to the psd function for normalization or pass in 2.5^2 as the rbw value (both hacks)

Well, the easiest, yes, but sadly incorrect. The 2.5dB holds only (and I guess only roughly) for white noise. When you have a clear tone, which is located only in a single bin (no spectral leakage), the current method will return the correct value. So this is a bit more complex than just multiplying with a constant.

>     * the average is an exponential decay - question: does the math still hold?

That does not matter. The implemented averaging is an efficient way of doing it, as it does not need to remember the past values (like a box averaging). Currently, this is just a simple 2-tap IIR filter, which just needs to remember the last value. Personally I don't like this kind of averaging in this case, as the response is infinite and it would be nice if the averaging would be "complete" in a known amount of time, but that is just my personal preference and has nothing to do with this problem ;)

>     * averaging was implemented from a graphics point of view for smoothing, so this was probably not considered

I can absolutely understand why someone did it like that, it is a very easy solution which requires just a little bit of code. And in most cases no-one will care about it (as apparently no-one has noticed this bug in years), yet, its wrong and I would appreciate a correct fix without a workaround - and @ali69550 probably as well. I have no clue how gnuradio works internally and I sadly don't have the time at the moment to look into it.

ali69550, 2021-07-11:
By the way! The WX GUI corresponding block does not show this issue.

marcusmueller, 2021-08-25:
So, there's no doubt here, the current method of calculation is wrong – but there *is* some beauty in it (in phosphorous displays you'd see a linear blending of the decibel-scale amplitudes, too). I'm not a fan of approximate displays myself, but I think when backporting this fix, it might be wise to add an option "decibel (visual) average" vs "actual power average" or so.

I'd be looking for suggestions on how to call these modes, actually!

marcusmueller, 2024-01-28:
hard to fix now without breaking a lot of expectations. Needs work!

argilo, 2025-03-28:
I've been annoyed by this too. The averaging gives very misleading results for signals where energy is intermittently present in the FFT bins. For instance, here's an HD Radio signal, where the analog FM in the center has 10 times the power of the OFDM sidebands. Without averaging:

[image]

With averaging:

[image]