gnuradio/gnuradio #4833: Segementation fault while running python-made blocks

araujoUnb, 2021-07-05:
I was trying to create the same square block implemented in python in the tutorial (https://wiki.gnuradio.org/index.php/OutOfTreeModules). However, when I ran the python test to check if the block is running as it should be, the python code returns a message indicating a segmentation fault.

Is there anyone who has experienced similar problems. My gnuradio is running on MacOs system, and I am using anaconda as a package manager.

Thanks in advance for the help.

willcode, 2021-07-09:
Leaving this open for now in case you've found a bug. More info would be needed for anyone to work on this, or even replicated. Recommend you close this issue and work with people on chat.gnuradio.org or the mailing list.

antonyscerri, 2021-10-03:
I've experienced the same issue using GNU Radio 3.9.2 on Mac. Installed into a conda env, everything for the most part works as expected. If I stretch the window too wide then it seems to crash doing various things due to some kind of bug in the QT GUI components. This specific issue though i built a simple general block with gr_modtool. I modified the input and output signatures replacing the placeholder strings. I've tried various input and output data types.

I created a single test case which simply does:

```
        src_data = (0,1,1,0,1,1,1,1,0)
        src = blocks.vector_source_b(src_data)
        comm = communicate()
        snk = blocks.vector_sink_b()

        self.tb.connect (src, comm)
        self.tb.connect (comm, snk)

        self.tb.run()
 ```

The block at this point has the default __init__ with in/out signatures set, and the general_work does nothing except consume the inputs and return the length of the output items. I've put various print statements in to see how far things went when running the test and its initialised the block. When the test calls:

```
self.tb.run()
```

Before the block's general_work method is called the segmentation fault occurs. I tried installing it anyway to see if had any better luck running inside and actual flowgraph. GNU Radio showed it (once i enabled it) and i could wire it up, but when the program was executed it failed with a segmentation fault.

I may go try a c++ version to see if that works.

willcode, 2021-10-03:
The flowgraph could be terminating before everything gets started. It still shouldn't segfault obviously.

What is the `communicate()` doing?

Try adding a `time.sleep(5), tb.stop(), tb.wait()` after `tb.run()` to see if it's a startup/shutdown problem.

antonyscerri, 2021-10-03:
Just gave that a quick try, no difference. It doesn't appear to be calling general_work. I got a c++ version going which at least starts running, but i have a separate issue there. Right now the communicate is not doing anything, i just copied the input so it had something. In C++ its going on forever and calling it with way more data than in the source provided and making many calls, but i can see its calling general_work at least.

Fairly new to GNU Radio and SDR in general. The aim, which i think may be a separate problem, was to have something reading input after demod and when appropriate send out data as bytes to then be modulated. I was hoping to in no way have any dependence on input items to output items (which forecast imposes), so it receives instructions and then sends them out when appropriate (in some cases it may want to send periodically regardless of input, which I now see will not actually be possible going by the api).

That all said it seems something in the pure python at least isnt working. I did note on the python tutorial page it does says that its deprecated as of 3.8.

willcode, 2021-10-03:
I'd design this so that `communitcate()` is in a separate thread and shares a queue/buffer with `work()`. That's `Popen.communicate()`? While there's no rule that `work()` can't have arbitary code in it, putting process contol code there seems like it could cause problems due to `work()` running in its own thread. That thread is different than the thread where the original `Popen` happened.

antonyscerri, 2021-10-03:
Sorry communicate here was just the name of my block. I was thinking for what I want to do, I need to create two flows one which receives transmissions and goes to a sync block, and then i create a second flow which has a source block in the same module which does my transmission flow. Then i just do what ever i need within the module to have my sink and source work in conjunction. I need to look more at sources and see whether they can simply send data as and when, rather than needing to generate a constant sample stream at a predefined rate.

On the segmentation fault front, i got gdb going on my mac, however launching it with the qa test script is just hanging so i'm still investigating this to see if I can get a stack trace or something to help narrow this seg fault down.

antonyscerri, 2021-10-03:
So gave up on gdb and just using python's faulthandler 👍 
```
Fatal Python error: Segmentation fault

Thread 0x0000000119b44dc0 (most recent call first):
  File ".../gnuradio/lib/python3.9/site-packages/gnuradio/gr/top_block.py", line 98 in start
  File ".../gnuradio/lib/python3.9/site-packages/gnuradio/gr/top_block.py", line 110 in run
  File ".../gr-module/python/qa_communicate.py", line 39 in test_001_descriptive_test_name
  File ".../gnuradio/lib/python3.9/unittest/case.py", line 550 in _callTestMethod
```

That last stack entry at line 98 is:

```
top_block_start_unlocked(self._impl, max_noutput_items)
```

willcode, 2021-10-04:
OK, then ignore my comment on communicate() ... the error may be in your block, since the code above looks pretty typical. Can you post that code?

antonyscerri, 2021-10-05:
I've attached the main communicate.py and the qa file i've been trying to run against it (sorry github didnt like .py extensions on attachments). Running it produces this output:

```
Test
Init
Ok
Ready
zsh: segmentation fault  python3 qa_communicate.py
```

[communicate.txt](https://github.com/gnuradio/gnuradio/files/7281945/communicate.txt)
[qa_communicate.txt](https://github.com/gnuradio/gnuradio/files/7281951/qa_communicate.txt)

willcode, 2021-10-05:
`basic_block` is an abstract class - try subclassing `block`.

Since we know Python blocks work, I'm going to close this issue. Please work with people on chat.gnuradio.org or the mailing list for further help getting your block to work.

closed 2021-10-05 by willcode

antonyscerri, 2021-10-05:
So the code i sent above is simply where i filled in the necessary bits after using gr_modtool to create the module and add a "general" block. The class definition is as it generated. So if that tool is creating incorrect code then maybe it needs changing. If you also look on this page https://wiki.gnuradio.org/index.php/Embedded_Python_Block in the given python code there is a comment which reads:

```
class blk(gr.sync_block):  # other base classes are basic_block, decim_block, interp_block
```

willcode, 2021-10-05:
You're right - I have no idea why modtool makes a basic_block for Python but a block for C++. When I generated the default Python block using modtool, it worked. I think you are required to implement forecast for a basic_block. The block class implements one for you. Modtool includes a forecast function for Python general blocks.

Also, consuming all the input items, as the generated code does, is not necessarily what you want, since the number of input items is not necessarily the same as the amount of output space given.

Most of the in-tree python blocks are not general ones, and the one you mention does not do any work.

So, go with a source/sink/sync/interp/decim if you can. Sorry about the confusion.

willcode, 2021-10-05:
One more piece of info ... the `block` class is exported as `basic_block` in `gateway.py`, probably a legacy naming issue. So, you were using a `block`. And it did work for me using the code generated by modtool, so maybe give it another try from scratch?

antonyscerri, 2021-10-05:
Would you mind posting the version which worked for you (the whole module folder if possible). I've tried it many times. Curious if you made a change to one of the other files.

willcode, 2021-10-05:
Current maint-3.9
```
gr_modtool newmod
cd gr-test
gr_modtool add testp
mkdir build
cd build
cmake -D<path to my install prefix> ..
make
make test
```
Module name is `test`, block name is `testp`, selected `general` for the Python block type, no params.

antonyscerri, 2021-10-05:
Ah....did you modify the test case and make use of the block? Following your steps its fine, but as soon as i fill in the empty test case with something like:

```
    def test_001_descriptive_test_name(self):
        # set up fg
        src_data = (0.0,1.0,1.0,0.0,1.0,1.0,1.0,1.0,0.0)
        src = blocks.vector_source_f(src_data)
        comm = testp()
        snk = blocks.vector_sink_f()

        self.tb.connect (src, comm)
        self.tb.connect (comm, snk)
        self.tb.run()
        # check data
```

That causes the seg fault.

willcode, 2021-10-05:
Alright, the autogenerated code is completely wrong. This works:

Block implementation
```
#!/usr/bin/env python
# -*- coding: utf-8 -*-
#
# Copyright 2021 gr-test author.
#
# SPDX-License-Identifier: GPL-3.0-or-later
#

import numpy
from gnuradio import gr

class testp(gr.basic_block):
    """
    docstring for block testp
    """
    def __init__(self):
        gr.basic_block.__init__(self,
            name="testp",
            in_sig=[numpy.float32, ],
            out_sig=[numpy.float32, ])

    def forecast(self, noutput_items, ninputs):
        #setup size of input_items[i] for work call
        return [noutput_items] * ninputs

    def general_work(self, input_items, output_items):
        n = min(len(input_items[0]), len(output_items[0]))
        output_items[0][:n] = input_items[0][:n]
        self.consume(0, n)
        return n
```
QA
```
#!/usr/bin/env python
# -*- coding: utf-8 -*-
#
# Copyright 2021 gr-test author.
#
# SPDX-License-Identifier: GPL-3.0-or-later
#

from gnuradio import gr, gr_unittest
from gnuradio import blocks
from testp import testp
import time

class qa_testp(gr_unittest.TestCase):

    def setUp(self):
        self.tb = gr.top_block()

    def tearDown(self):
        self.tb = None

    def test_instance(self):
        # FIXME: Test will fail until you pass sensible arguments to the constructor
        src_data = (0.0,1.0,1.0,0.0,1.0,1.0,1.0,1.0,0.0)
        src = blocks.vector_source_f(src_data, False)
        instance = testp()
        snk = blocks.vector_sink_f()
        self.tb.connect(src, instance)
        self.tb.connect(instance, snk)
        print('run')
        self.tb.run()
        time.sleep(2)
        self.tb.stop()
        self.tb.wait()

    def test_001_descriptive_test_name(self):
        # set up fg
        self.tb.run()
        # check data
```

willcode, 2021-10-05:
See issue https://github.com/gnuradio/gnuradio/issues/5143