gnuradio/gnuradio #4834: Core -> Trellis Coding: Some blocks fail to import c bindings

mcdiarmid, 2021-07-05:
Any flowgraph containing one of the following blocks will raise an AttributeError:
* Trellis Metrics
* Viterbi Combo
* SCCC Decoder
* SCCC Decoder Combo
* SCCC Encoder
* PCCC Decoder
* PCCC Decoder Combo
* PCCC Decoder

An example with the Trellis Metrics block:
`AttributeError: module 'gnuradio.trellis' has no attribute 'metrics_c'`

It appears that the python bindings either cannot be found, or the code for these blocks hasn't been installed.
```python
>>> from gnuradio import trellis
>>> dir(trellis)
['TRELLIS_MIN_SUM', 'TRELLIS_SUM_PRODUCT', '__builtins__', '__cached__', '__doc__', 
'__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', 
'constellation_metrics_cf', 'encoder_bb', 'encoder_bi', 'encoder_bs', 'encoder_ii', 
'encoder_si', 'encoder_ss', 'fsm', 'interleaver', 'os', 'permutation', 'siso_combined_f', 
'siso_f', 'siso_type_t', 'trellis_python', 'viterbi_b', 'viterbi_i', 'viterbi_s']
```

I've experienced this issue in GNURadio 3.7, 3.8 and 3.9.  I'm assuming this is due to a bad install of GNURadio, or that these blocks were developed in an older version of GNURadio and have succumb to backwards compatibility issues - I suspect it is the latter as I have tried installing various installs across multiple versions and systems.  If there is a quick work-around for this I would greatly appreciate guidance on the matter! 

I also notice that examples and tests for blocks in gr-trellis are out of date.  For example, `examples/trellis/tcm.grc` attempts to use a missing block for calculating BER.

mcdiarmid, 2021-07-05:
This must be directly related to #2297 given gr-trellis hasn't been checked off.  I'm going to try manually construct the binding files and share them here if the workaround is successful

dkozel, 2021-07-06:
Hi @mcdiarmid Are you trying to generate a C++ application from GNU Radio Companion? By default it produces a Python application. #2297 is only relevant if you're generating a C++ app.

I haven't checked that it runs correctly, but on my GNU Radio 3.8 install the gr-trellis pccc example does run once the missing BER block is removed. Python bindings changed between 3.8 and 3.9. Can you test 3.9 and let us know what OS you are on and how you have installed GNU Radio?

The fastest place for live debugging help is at https://chat.gnuradio.org if you're able to join there. Otherwise we can work through things here until it's confirmed this isn't a bug in GNU Radio.

mcdiarmid, 2022-07-14:
This issue was addressed in #5707, and 7 blocks in gr-trellis now have their bindings!

However, [metrics](https://github.com/gnuradio/gnuradio/blob/main/gr-trellis/python/trellis/bindings/metrics_python.cc) and [quicksort_index](https://github.com/gnuradio/gnuradio/blob/main/gr-trellis/python/trellis/bindings/quicksort_index_python.cc) appear to have been missed.  This seems to be getting tracked in #5895, so might be okay to close this issue out.

closed 2022-07-17 by mcdiarmid