gnuradio/gnuradio #4841: ImportError: generic_type: type "square_ff" referenced unknown base type "gr::block"

wsx000, 2021-07-07:
i met a bug when i do a OOT test followed https://wiki.gnuradio.org/index.php/OutOfTreeModules, it says:
```
wsx@wsx:~/temp/gr-mymodule/build$ ctest -V -R square
UpdateCTestConfiguration  from :/home/wsx/temp/gr-mymodule/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/wsx/temp/gr-mymodule/build/DartConfiguration.tcl
Test project /home/wsx/temp/gr-mymodule/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: qa_square_ff

1: Test command: /usr/bin/sh "/home/wsx/temp/gr-mymodule/build/python/qa_square_ff_test.sh"
1: Test timeout computed to be: 10000000
1: Traceback (most recent call last):
1:   File "/home/wsx/temp/gr-mymodule/python/qa_square_ff.py", line 12, in <module>
1:     from mymodule import square_ff
1:   File "/usr/local/lib/python3/dist-packages/mymodule/__init__.py", line 18, in <module>
1:     from .mymodule_python import *
1: ImportError: generic_type: type "square_ff" referenced unknown base type "gr::block"
1: 
1: During handling of the above exception, another exception occurred:
1: 
1: Traceback (most recent call last):
1:   File "/home/wsx/temp/gr-mymodule/python/qa_square_ff.py", line 18, in <module>
1:     from mymodule import square_ff
1:   File "/usr/local/lib/python3/dist-packages/mymodule/__init__.py", line 18, in <module>
1:     from .mymodule_python import *
1: ImportError: generic_type: type "square_ff" referenced unknown base type "gr::block"
1/1 Test #1: qa_square_ff .....................***Failed    0.18 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.18 sec

The following tests FAILED:
	  1 - qa_square_ff (Failed)
Errors while running CTest
```
and my GR's version is 3.9.2, can anyone help me?

mormj, 2021-07-08:
https://www.mail-archive.com/discuss-gnuradio@gnu.org/msg73247.html

See if this fixes your problem - if GNU Radio was installed via the ppa on launchpad, this has previously caused a similar issue

willcode, 2021-07-25:
@wsx000 Did the `apt upgrade pybind11-dev` recommended by @mormj work for you?

wsx000, 2021-09-03:
> @wsx000 Did the `apt upgrade pybind11-dev` recommended by @mormj work for you?

apt upgrade pybind11-dev commend cant upgrade my pybind11-dev to v2.5.0, i dont know why.  so i converted my gnuradio to v3.8 in the end.

willcode, 2021-09-03:
Solved (and this was a help request, not a bug anyway).

closed 2021-09-03 by willcode

wsx000, 2021-11-17:
thanks，i solved this problem by i install GR 3.9.4