gnuradio/gnuradio #4842: from 3.7 to 3.8?

liumaolin-931, 2021-07-07:
Hi,
I am studying the file transfer process recently, I found an open source project https://github.com/BogdanDIA/gr-dvbt, I want to install and use it, but it is built on version 3.7, my gnuradio version is 3.8.2.0, when I mkdir build and execute cmake, there is no unexpected error, so how should I install it to the existing 3.8 version correctly? I took a look at the official tutorial, but it’s really too short and I don’t understand how to do it. Can anyone help me?

drmpeg, 2021-07-07:
All of gr-dvbt has been integrated into GNU Radio. Take a look in the Digital Television category of GNU Radio Companion. The example flow graphs are in <install_directory>/share/gnuradio/examples/dtv.

You can find your install directory with ```gnuradio-config-info --prefix```

[image]

willcode, 2021-07-07:
Help request - closing. Please use chat.gnuradio.org or the mailing list if you need additional help.

closed 2021-07-07 by willcode

liumaolin-931, 2021-07-08:
thanks a lot!