gnuradio/gnuradio #4845: gnuradio 3.10 - Crash

Phil67ago, 2021-07-07:
Gnuradio exit suddenly.
- In a new Flowgraph add a 'Python Block' 
- Properties and 'Open in Editor' and insert following code:
- In editor 'save'.
- Now gnuradio exit with console text:

**QWidget: Must construct a QApplication before a QWidget**

Code:

`from PyQt5.QtWidgets import QFrame, QHBoxLayout, QVBoxLayout, QLabel
from PyQt5.QtCore import Qt as Qtc
from gnuradio import gr
import pmt
import numpy as np

class LabelField(QFrame):
    def __init__(self, parent=None):
        QFrame.__init__(self, parent)
        self.show()

class blk(gr.sync_block, LabelField):

    def __init__(self, example_param=1.0):
        gr.sync_block.__init__( self,  name='Embedded Python Block', in_sig=None, out_sig=None )
        LabelField.__init__(self)
        self.example_param = example_param

        self.message_port_register_in(pmt.intern("state"))
        self.set_msg_handler(pmt.intern("state"), self.msgHandler)

    def work(self, input_items, output_items):
        return 0`

dkozel, 2021-07-07:
I'm just headed to bed, but wanted to link your related mailing list post here for extra context.
https://lists.gnu.org/archive/html/discuss-gnuradio/2021-07/msg00035.html

> Hi!
> 
> How do I do when I need to update 'QT GUI Label' block from inside of 'Embedded 
> Python Block'.
> I like to take a calculated value from _work and update a GUI label on screen 
> with the result.
> Any Example....
> 
> I'm running gnuradio.3-10 master.

The error message is telling you what the problem is, that the block is using QT elements before the overall QTApplication has been setup. I don't have any specific advice except that you should look at the implementation of the existing QT GUI blocks and see how they setup the QT code during initialization.

Phil67ago, 2021-07-08:
Hi! 

Yes documentation definitely need an update!

I was little concerned about the sudden exit.
If your work is not saved when this happens everything is lost.
If it is possible to catch that and not exit would be good!

/Thanks 
Phil

marcusmueller, 2024-01-29:
Not really something we can fix; this is really standard Qt.

closed 2024-01-29 by marcusmueller