gnuradio/gnuradio #4862: Probe Signal / Function Probe Broken with some Qt GUI Blocks

I-AM-EXFOLIATING, 2021-07-11:
[image]
<details>
<summary>
probe_test.grc
</summary>

```options:
  parameters:
    author: xfol
    catch_exceptions: 'True'
    category: '[GRC Hier Blocks]'
    cmake_opt: ''
    comment: ''
    copyright: ''
    description: Test Probe Signal Block
    gen_cmake: 'On'
    gen_linking: dynamic
    generate_options: qt_gui
    hier_block_src_path: '.:'
    id: probe_test
    max_nouts: '0'
    output_language: python
    placement: (0,0)
    qt_qss_theme: ''
    realtime_scheduling: ''
    run: 'True'
    run_command: '{python} -u {filename}'
    run_options: prompt
    sizing_mode: fixed
    thread_safe_setters: ''
    title: Probe Signal Test
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [8, 8]
    rotation: 0
    state: enabled

blocks:
- name: ampl
  id: variable_function_probe
  parameters:
    block_id: ampl_probe
    comment: ''
    function_args: ''
    function_name: level
    poll_rate: '100'
    value: '0'
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [440, 124.0]
    rotation: 0
    state: true
- name: qtgui_dialgauge_0
  id: qtgui_dialgauge
  parameters:
    affinity: ''
    alias: ''
    backgroundColor: white
    barColor: blue
    comment: ''
    fontColor: black
    gui_hint: ''
    label: Amplitude
    maximum: '2'
    minimum: '0'
    msize: '100'
    position: '1'
    showValue: 'True'
    type: real
    value: ampl
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [608, 204.0]
    rotation: 0
    state: disabled
- name: qtgui_levelgauge_0
  id: qtgui_levelgauge
  parameters:
    affinity: ''
    alias: ''
    backgroundColor: default
    barColor: default
    comment: ''
    fontColor: default
    gui_hint: ''
    isVertical: 'True'
    label: Amplitude
    maximum: '100'
    minimum: '0'
    msize: '100'
    position: '1'
    scaleFactor: '1'
    showValue: 'False'
    type: int
    value: ampl
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [608, 396.0]
    rotation: 0
    state: disabled
- name: samp_rate
  id: variable
  parameters:
    comment: ''
    value: '32000'
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [184, 12]
    rotation: 0
    state: enabled
- name: variable_qtgui_label_0
  id: variable_qtgui_label
  parameters:
    comment: ''
    formatter: None
    gui_hint: ''
    label: Amplitude
    type: real
    value: ampl
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [600, 124.0]
    rotation: 0
    state: enabled
- name: ampl_probe
  id: blocks_probe_signal_x
  parameters:
    affinity: ''
    alias: ''
    comment: ''
    type: float
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [456, 220.0]
    rotation: 0
    state: enabled
- name: analog_sig_source_x_0
  id: analog_sig_source_x
  parameters:
    affinity: ''
    alias: ''
    amp: '1'
    comment: ''
    freq: '1000'
    maxoutbuf: '0'
    minoutbuf: '0'
    offset: '0'
    phase: '0'
    samp_rate: samp_rate
    type: float
    waveform: analog.GR_COS_WAVE
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [104, 180.0]
    rotation: 0
    state: true
- name: blocks_throttle_0
  id: blocks_throttle
  parameters:
    affinity: ''
    alias: ''
    comment: ''
    ignoretag: 'True'
    maxoutbuf: '0'
    minoutbuf: '0'
    samples_per_second: samp_rate
    type: float
    vlen: '1'
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [296, 220.0]
    rotation: 0
    state: true
- name: qtgui_ledindicator_0
  id: qtgui_ledindicator
  parameters:
    affinity: ''
    alias: ''
    cellalignment: '1'
    comment: ''
    gui_hint: ''
    label: '"Amplitude"'
    maxSize: '40'
    offColor: red
    onColor: green
    position: '1'
    state: ampl
    verticalalignment: '1'
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [600, 324.0]
    rotation: 0
    state: disabled

connections:
- [analog_sig_source_x_0, '0', blocks_throttle_0, '0']
- [blocks_throttle_0, '0', ampl_probe, '0']

metadata:
  file_format: 1
```
</details>

When the flowgraph above is executed, the UI displays values from the Probe Signal correctly on the QT GUI Label, however if one of the QT GUI Dial Gauge / LED Indicator / Level Gauge blocks^ are enabled, in combination or isolation, then a traceback is output as follows:
^ Other blocks may be affected but these are the ones I've tried.
```
Executing: /usr/bin/python3 -u /home/bryce/Documents/probe_test/probe_test.py

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/bryce/Documents/probe_test/probe_test.py", line 91, in _ampl_probe
    val = self.ampl_probe.level()
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 82, in __getattr__
    return getattr(self._impl, name)
AttributeError: 'gnuradio.gr.gr_python.top_block_pb' object has no attribute 'ampl_probe'
```

The declaration for `ampl_probe` exists in the generated Python code but its declaration is deferred until some time after the declaration of the closure that references it (`_ampl_probe`).

The generated code which works:
<pre><code>  4         <b>self.ampl_probe = blocks.probe_signal_f()</b>
  3         def _ampl_probe():
  2           while True:
  1 
103             <b>val = self.ampl_probe.level()</b>
  1             try:
  2               try:
  3                 self.doc.add_next_tick_callback(functools.partial(self.set_ampl,val))
  4               except AttributeError:
  5                 self.set_ampl(val)
  6             except AttributeError:
  7               pass
  8             time.sleep(1.0 / (100))
  9         _ampl_thread = threading.Thread(target=_ampl_probe)
</code></pre>

And the generated code which doesn't work:
<pre><code>  3         def _ampl_probe():
  2           while True:
  1 
89              <b>val = self.ampl_probe.level()</b>
  1             try:
  2               try:
  3                 self.doc.add_next_tick_callback(functools.partial(self.set_ampl,val))
  4               except AttributeError:
  5                 self.set_ampl(val)
  6             except AttributeError:
  7               pass
  8             time.sleep(1.0 / (100))
  9         _ampl_thread = threading.Thread(target=_ampl_probe)
 10         _ampl_thread.daemon = True
 11         _ampl_thread.start()
 12         self.qtgui_ledindicator_0 = self._qtgui_ledindicator_0_win = qtgui.GrLEDIndicator("Amplitude", "green", "red", ampl, 40, 1,     1, 1, self)
 13         self.qtgui_ledindicator_0 = self._qtgui_ledindicator_0_win
 14         self.top_layout.addWidget(self._qtgui_ledindicator_0_win)
 15         self.blocks_throttle_0 = blocks.throttle(gr.sizeof_float*1, samp_rate,True)
 16         self.analog_sig_source_x_0 = analog.sig_source_f(samp_rate, analog.GR_COS_WAVE, 1000, 1, 0, 0)
 17         <b>self.ampl_probe = blocks.probe_signal_f()</b>
</code></pre>
Note the relative ordering of `self.ampl_probe = blocks.probe_signal_f()` and `val = self.ampl_probe.level()` (in bold).

willcode, 2021-07-11:
Possibly related to https://github.com/gnuradio/gnuradio/pull/4695? @ChristopheSeguinot what do you think?

oe1rsa, 2023-01-04:
I am also affected by this bug, however, the solution suggested by the original poster requires a reordering of other blocks which is harder to achieve than the following suggested solution which needs only a local change:

Just move the line before the first try, i.e.
     val = self.ampl_probe.level()
behind the the first try, i.e. **into** the outer try block.

Anyways, perhaps the position of the val = line originally was intended to be after the first try, because the code as it currently is does not make a lot of sense to me because the outer try block looks meaningless as it currently is.

willcode, 2023-01-04:
Agree that the code is generated out of order, same problem here. @haakov or others - any ideas? This is a pretty old one, and may have been cause by attempts to emit code in the correct order to fix other bugs.

haakov, 2023-01-07:
Aha, I'll take a look at it tomorrow

haakov, 2023-01-08:
Interesting. I'm able to reproduce with current main.

haakov, 2023-01-08:
I wasn't able to get all the way to the bottom of it tonight, but there's a problem with the output of `sort_objects()` in `grc/core/utils/expr_utils.py`. 

More specifically, this line:
https://github.com/gnuradio/gnuradio/blob/c30aef2e7dc365cdb35de50bd57f977899ffb747/grc/core/generator/top_block.py#L192
causes the expression for the Function Probe to not include any reference to the Probe Signal block, and thus the `make` section  for the Probe Signal block is placed after the Function Probe in the generated Python code.

haakov, 2023-01-10:
Looks like that's correct. Renaming the Probe Signal block to something that doesn't contain "ampl" works on my end

oe1rsa, 2023-01-13:
Is there a workaround for us mere mortals? Putting the code within a Python Snippet unfortunately doesn't work.
Going to an external editor, moving the offending line into the first try block, however, works. Of course I have to start the script from the commandline then.

Sadly I do not see this as a work-around.

oe1rsa, 2023-01-13:
Oh, mistake on my side: The work-around of putting the code into a python snippet does work when one does it right: i.e. not forget to include threading and time.
So for the moment "it works for me". Hopefully you will be able to find the bug and correct it.
Thank you very much for your efforts!

haakov, 2023-01-13:
There is a workaround! :)

Choose a different ID for the Probe Signal block. Any ID that does not contain "ampl" is fine, so "a_probe" or "probe_sig" will work.

marcusmueller, 2024-01-29:
yeah, bug. Note we should really deprecate these blocks; almost always a bad idea. Where they shouldn't actually be replaced with decimators (which almost always is the case), a message-based workflow would be adequate and safe.