gnuradio/gnuradio #4868: Constant Source value 'short' not in allowed values v3.8.3.1

lmr12, 2021-07-12:
I recently installed v3.8.3.1 via pybombs, and now when I try to run flowgraphs which use the Constant Source block with the Output Type "short" the properties tab show that "value 'short' no in allowed values". This was not an issue in v3.8.2.0 and I am wondering if this was an intentional change. I did not see any mention in the change-log.
[image]
[image]
[image]

willcode, 2021-07-12:
@ChristopheSeguinot @dl1ksv something related to type checking and/or ordering, I think.

Also affects master and maint-3.9. `byte` does not work either. Other blocks allow `short` and `byte`.

In `gr-analog/grc/analog_const_source_x.block.yml`:
```
parameters:
-   id: type
    label: Output Type
    dtype: enum
    options: [complex, float, int, short, byte]
    option_attributes:
        const_type: [complex, real, int, short, gr.sizeof_char]
        fcn: [c, f, i, s, b]
    hide: part
-   id: const
    label: Constant
    dtype: ${ type.const_type }
    default: '0'

outputs:
-   domain: stream
    dtype: ${ type }
```
Adding a print to `EvaluatedEnum(Evaluated)` show
```
Value 'gr.sizeof_char' not in allowed values {'_multiline_python_external', 'complex_vector', 'enum', 'id', 'real', 'int_vector', 'gui_hint', 'int', 'float_vector', 'file_open', 'stream_id', 'string', 'import', 'hex', 'dir_select', 'complex', 'real_vector', 'file_save', 'raw', 'float', '_multiline', 'name', 'bool'}
```
which is clearly not type checking the right enum.

willcode, 2021-07-13:
This gets it part way fixed:
```
$ git diff
diff --git a/gr-analog/grc/analog_const_source_x.block.yml b/gr-analog/grc/analog_const_source_x.block.yml
index 929ed6496..3fdb883d7 100644
--- a/gr-analog/grc/analog_const_source_x.block.yml
+++ b/gr-analog/grc/analog_const_source_x.block.yml
@@ -8,7 +8,7 @@ parameters:
     dtype: enum
     options: [complex, float, int, short, byte]
     option_attributes:
-        const_type: [complex, real, int, short, gr.sizeof_char]
+        const_type: [complex, real, int, short, byte]
         fcn: [c, f, i, s, b]
     hide: part
 -   id: const
diff --git a/grc/core/Constants.py b/grc/core/Constants.py
index e7d8f3f99..2d408aea4 100644
--- a/grc/core/Constants.py
+++ b/grc/core/Constants.py
@@ -58,7 +58,7 @@ HIER_BLOCK_FILE_MODE = stat.S_IRUSR | stat.S_IWUSR | stat.S_IRGRP | stat.S_IWGRP
 
 PARAM_TYPE_NAMES = {
     'raw', 'enum',
-    'complex', 'real', 'float', 'int',
+    'complex', 'real', 'float', 'int', 'short', 'byte',
     'complex_vector', 'real_vector', 'float_vector', 'int_vector',
     'hex', 'string', 'bool',
     'file_open', 'file_save', '_multiline', '_multiline_python_external',
diff --git a/grc/core/params/param.py b/grc/core/params/param.py
index b3570b946..73ad151f7 100644
--- a/grc/core/params/param.py
+++ b/grc/core/params/param.py
@@ -218,7 +218,7 @@ class Param(Element):
         #########################
         # Numeric Types
         #########################
-        elif dtype in ('raw', 'complex', 'real', 'float', 'int', 'hex', 'bool'):
+        elif dtype in ('raw', 'complex', 'real', 'float', 'int', 'short', 'byte', 'hex', 'bool'):
             if expr:
                 try:
                     if isinstance(expr, str) and self.is_float(expr[:-1]):
```

dl1ksv, 2021-07-13:
@willcode Your patch works for me on the master branch for both: short and byte.
Did youtry to clear .cache/grc_gnuradio/* ?

willcode, 2021-07-13:
Yes, it works. My environment variables got tangled up and it seemed to be magically regenerating. I'll put this in as a PR.

willcode, 2021-07-13:
Still, it seems like the enum check should be against the valid values for the enum, rather than the global set of valid types, doesn't it?

willcode, 2021-07-13:
@lmr12 We'll get this fixed, but it won't be in an offiical release until 9/2021. In the mean time, you can
- Use Vector Source instead of Constant Source, OR
- Edit the Python and/or YAML files on your system to match the diff above

lmr12, 2021-07-13:
@willcode thank you.

closed 2021-07-19 by mormj