gnuradio/gnuradio #4874: can't Execut all GRC AttributeError: 'module' object has no attribute 'QWidget'

zjklxw, 2021-07-13:
When I execut a simple GRC project in gnuradio-companion, it can be  executed normally. The GRC flowgraph would generate a python file (top_block.py). If I use python2.7
$ python top_block.py , an error occured as followed:

Executing: /opt/local/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u /Users/cn0xroot/top_block.py

Traceback (most recent call last):
  File "/Users/cn0xroot/top_block.py", line 29, in <module>
    from gnuradio import qtgui
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/qtgui/__init__.py", line 37, in <module>
    from range import Range, RangeWidget
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/qtgui/range.py", line 67, in <module>
    class RangeWidget(QtGui.QWidget):
AttributeError: 'module' object has no attribute 'QWidget'

This error won't happen when I execute the grc flowgraph.

dkozel, 2021-07-13:
Hi @zjklxw 

When you run the flowgraph from GRC it will print the full Python execution command in the console in the bottom left of the GRC window. Does running exactly that command work? You are probably using a different Python environment. Recent GNU Radio versions also only support Python 3, though that's not what this error looks like.

zjklxw, 2021-07-14:
> Hi @zjklxw
>
> When you run the flowgraph from GRC it will print the full Python execution command in the console in the bottom left of the GRC window. Does running exactly that command work? You are [...]

Thank you for your reply!

I worked in Gnuradio 3.7.11, which supports python 2.7 and I execute the .py file in python 2.7 . I don't think the problem is caused by different python interpreter. It is quite strange. I reinstalled all my ubuntu and tried the same operation, it worked...

and the other python file which I modified according to a GRC flowgraph also worked.  It shouldn't be a problem...

closed 2021-07-14 by zjklxw