gnuradio/gnuradio #4875: Error: Can't parse input/output signature.

wsx000, 2021-07-13:
i meet a strange problem ! though i have resolved this problem, i still fill magical and puzzled ! 
as the topic says, when i write my owen block, and use the commend line 
`gr_modtool makeyaml square1_ff`
 to general yaml file, it reported error at the first time
`wsx@wsx:~/temp/gr-mymod$ gr_modtool makeyaml square1_ff
GNU Radio module name identified: mymod
Warning: This is an experimental feature. Don't expect any magic.
Searching for matching files in lib/:
Making GRC bindings for lib/square1_ff_impl.cc...
Error: Can't parse input signature.
Error: Can't parse output signature.
Traceback (most recent call last):
  File "/usr/local/bin/gr_modtool", line 30, in <module>
    cli()
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3/dist-packages/gnuradio/modtool/cli/base.py", line 153, in wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python3/dist-packages/gnuradio/modtool/cli/makeyaml.py", line 48, in cli
    run(self)
  File "/usr/local/lib/python3/dist-packages/gnuradio/modtool/cli/base.py", line 172, in run
    module.run()
  File "/usr/local/lib/python3/dist-packages/gnuradio/modtool/core/makeyaml.py", line 69, in run
    self._make_grc_yaml_from_block_data(params, iosig, blockname)
  File "/usr/local/lib/python3/dist-packages/gnuradio/modtool/core/makeyaml.py", line 93, in _make_grc_yaml_from_block_data
    if iosig[inout]['max_ports'] == '-1':
KeyError: 'in'
`
and i tried many methods to fix but it always didn't work... 
Unintentionally , i find that when i write the c++ code like this
`    square1_ff_impl::square1_ff_impl()
      : gr::block("square1_ff",
              gr::io_signature::make(1, 1, sizeof(float)),   // whatever annotation
              gr::io_signature::make(1, 1, sizeof(float)))   // whatever annotation
    {}`
the commend line did work and will report error as mentioned in this topic.
but when i write code like this
`    square1_ff_impl::square1_ff_impl()
      : gr::block("square1_ff",
              gr::io_signature::make(1, 1, sizeof(float)),   
              gr::io_signature::make(1, 1, sizeof(float)))   
    {}`
the commend line work normally！
do you find this different? yes, the only diferencd is the existence of annotation behind the sentence 
`gr::io_signature::make(1, 1, sizeof(float))`
Does anyone know why？

mormj, 2021-07-14:
The code that tries to parse the io_signature uses some limited regex to obtain the number and type of ports that should be declared in the yaml.  Apparently it does not handle the case when there is a line comment in the middle of the expression.

wsx000, 2021-07-15:
> The code that tries to parse the io_signature uses some limited regex to obtain the number and type of ports that should be declared in the yaml. Apparently it does not handle the case when there is a line comment in the middle of the expression.

It explains the matter，thanks！

closed 2021-07-15 by wsx000