gnuradio/gnuradio #4891: Failed to start GRC

mooction, 2021-07-15:
After compiled gnuradio (branch `maint-3.9`), I failed to start grc, it show me a message, and the program exited:

> Failed to initialize GTK. If you are running over ssh......

Then I edit `/usr/local/bin/gnuradio-companion`:

```py
def check_gtk():
    try:
        # ========REMOVE THIS LINE==========
        #warnings.filterwarnings("error")
        import gi
        gi.require_version('Gtk', '3.0')
        gi.require_version('PangoCairo', '1.0')
        gi.require_foreign('cairo', 'Context')

        from gi.repository import Gtk
        Gtk.init_check()
        warnings.filterwarnings("always")
    except Exception as err:
        die(err, "Failed to initialize GTK. If you are running over ssh, "
                 "did you enable X forwarding and start ssh with -X?")
```

And it works, the output:

```
/usr/lib/python3/dist-packages/gi/overrides/Gdk.py:461: Warning: g_value_set_boxed: assertion 'G_VALUE_HOLDS_BOXED (value)' failed
  initialized, argv = Gdk.init_check(sys.argv)
<<< Welcome to GNU Radio Companion v3.9.2.0-50-g1a4bbabb >>>

Block paths:
        /home/fang/.grc_gnuradio
        /usr/local/share/gnuradio/grc/blocks
```

I think i missed something. Dose anyone know it?

dkozel, 2021-07-15:
Hi @mooction

What operating system are you compiling on?

mooction, 2021-07-16:
@dkozel Deepin 20.2.1 (upstream Debian version 10.9)

`cmake -L ../`

```text
-- Build type set to Release.
-- Build date set to Fri, 16 Jul 2021 01:44:22Z.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- 
-- Configuring testing-support support...
--   Dependency Boost_UNIT_TEST_FRAMEWORK_FOUND = ON
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
-- Extracting version information from git describe...
-- Compiler Version: cc (Uos 8.3.0.3-3+rebuild) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- User set python executable /usr/bin/python3
-- Python checking for numpy - found
-- Python checking for pygccxml - not found
-- Found pybind11: /usr/local/include (found version "2.6.3" dev1)
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency pybind11_FOUND = 1
--   Dependency NUMPY_FOUND = TRUE
--   Enabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- 
-- Configuring VOLK support...
--   Found VOLK:
--   * Version: 2.5.0
--   * Libraries: Volk::volk
--   * Includes: /usr/local/include
-- Not using additional GNU Radio native architecture optimizations.
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Using GMP.
-- 
-- Configuring post-install support...
--   Enabling post-install support.
--   Override with -DENABLE_POSTINSTALL=ON/OFF
-- Could NOT find MathJax2 (missing: MATHJAX2_JS_PATH) 
-- 
-- Configuring doxygen support...
--   Dependency DOXYGEN_FOUND = YES
--   Enabling doxygen support.
--   Override with -DENABLE_DOXYGEN=ON/OFF
CMake Warning at docs/doxygen/CMakeLists.txt:24 (message):
  MathJax 2 not found, HTML equations might not be properly rendered.

-- Checking for module 'gmp'
--   No package 'gmp' found
[... 371 lines cut ...]
SHELL:FILEPATH=/usr/bin/sh
SYSCONFDIR:PATH=/usr/local/etc
SoapySDR_DIR:PATH=SoapySDR_DIR-NOTFOUND
THRIFT_BIN:FILEPATH=THRIFT_BIN-NOTFOUND
THRIFT_INCLUDE_DIRS:PATH=THRIFT_INCLUDE_DIRS-NOTFOUND
THRIFT_LIBRARIES:FILEPATH=THRIFT_LIBRARIES-NOTFOUND
TRY_SHM_VMCIRCBUF:BOOL=ON
UHD_DIR:PATH=/usr/lib/x86_64-linux-gnu/cmake/uhd
VOLK_FOUND:BOOL=TRUE
Volk_DIR:PATH=/usr/local/lib/cmake/volk
XMLTO_EXECUTABLE:FILEPATH=XMLTO_EXECUTABLE-NOTFOUND
libunwind_INCLUDE_DIR:PATH=libunwind_INCLUDE_DIR-NOTFOUND
libunwind_LIBRARY:FILEPATH=libunwind_LIBRARY-NOTFOUND
pkgcfg_lib_ALSA_PKGCONF_asound:FILEPATH=/usr/lib/x86_64-linux-gnu/libasound.so
pkgcfg_lib_GSL_gsl:FILEPATH=/usr/lib/x86_64-linux-gnu/libgsl.so
pkgcfg_lib_GSL_gslcblas:FILEPATH=/usr/lib/x86_64-linux-gnu/libgslcblas.so
pkgcfg_lib_GSL_m:FILEPATH=/usr/lib/x86_64-linux-gnu/libm.so
pkgcfg_lib_LIBCODEC2_PKG_codec2:FILEPATH=/usr/lib/x86_64-linux-gnu/libcodec2.so
pkgcfg_lib_PC_FFTW3f_fftw3f:FILEPATH=/usr/lib/x86_64-linux-gnu/libfftw3f.so
pkgcfg_lib_PC_SNDFILE_sndfile:FILEPATH=/usr/lib/x86_64-linux-gnu/libsndfile.so
pkgcfg_lib_PC_UHD_boost_system:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so
pkgcfg_lib_PC_UHD_uhd:FILEPATH=/usr/lib/x86_64-linux-gnu/libuhd.so
pkgcfg_lib_PC_ZEROMQ_zmq:FILEPATH=/usr/lib/x86_64-linux-gnu/libzmq.so
pybind11_DIR:PATH=/usr/local/share/cmake/pybind11
```

marcusmueller, 2021-07-16:
So, there's two things at work here:

1. GTK/GDK on your machine is not happy with itself. In the `Gtk.init_check()` line, it does check whether its internal typing system is in good order, and find it should be emitting a warning. So, that's not something that GNU Radio can fix!
2. GNU Radio might be a bit overly sensitive here: A warning is a warning, not something fatal. But I **bet** that conversion of warnings to exceptions (which you commented out) is in there for a reason! So we need to figure out that reason.

marcusmueller, 2021-07-16:
@mooction can you change things to

```python
def check_gtk():
    try:
        # ========CHANGE THIS LINE==========
        warnings.filterwarnings("always")
        import gi
        gi.require_version('Gtk', '3.0')
        gi.require_version('PangoCairo', '1.0')
        gi.require_foreign('cairo', 'Context')

        from gi.repository import Gtk
        if not Gtk.init_check():
            warnings.warn("init_check unsuccessful. We're in a bad state. Every usage of Gtk after this point has undefined behaviour!\nhttps://developer.gnome.org/gtk3/stable/gtk3-General.html#gtk-init-check")
        warnings.filterwarnings("always")
    except Exception as err:
        die(err, "Failed to initialize GTK. If you are running over ssh, "
                 "did you enable X forwarding and start ssh with -X?")
```

and tell us the console output of starting gnuradio-companion?

mooction, 2021-07-20:
@marcusmueller I used the code, and GRC works. The output is the same:

```text
/usr/lib/python3/dist-packages/gi/overrides/Gdk.py:461: Warning: g_value_set_boxed: assertion 'G_VALUE_HOLDS_BOXED (value)' failed
  initialized, argv = Gdk.init_check(sys.argv)
<<< Welcome to GNU Radio Companion v3.9.2.0-50-g1a4bbabb >>>

Block paths:
        /home/fang/.grc_gnuradio
        /usr/local/share/gnuradio/grc/blocks
```

Looks not so bad?

mormj, 2022-01-20:
@marcusmueller - can you verify that this can be closed based on #5216 and #5217

marcusmueller, 2024-01-29:
Yep, seems done.

closed 2024-01-29 by marcusmueller