gnuradio/gnuradio #4892: Constellation power normalization and incorrect soft decision LUT for M-QAM with M > 4

jfmadeira, 2021-07-15:
Hello all,

I have a question about the constellation power normalization in [constellation.cc](https://github.com/gnuradio/gnuradio/blob/master/gr-digital/lib/constellation.cc). If using QPSK with power normalization (lines 53-63), the resulting constellation will contain (+-sqrt(2) +-sqrt(2)*j). Doesn't this constellation have an average power of 4?

Secondly, I believe the soft decision LUT generation in this file is incorrect for constellations greater QPSK, because it assumes the constellation is normalized by the max amplitude, and not the average amplitude (line 246), which in the case of QPSK are the same.

dave-pi, 2021-07-15:
Regarding the first question I think you're right (second example is what the code is doing):

```py
>>> import numpy as np
>>> num = 4 # a random number of points
>>> a = np.random.rand(num) + np.random.rand(num)*1j
>>> average_power = np.sum(abs(a)**2)/len(a)
>>> a_normalized = a/sqrt(average_power)
>>> np.sum(abs(a_normalized)**2)/len(a_normalized) # average power of normalized points
1.0
>>>
>>> import numpy as np
>>> num = 4 # a random number of points
>>> a = np.random.rand(num) + np.random.rand(num)*1j
>>> average_power = np.sum(abs(a)**2)
>>> a_normalized = a*len(a)/sqrt(average_power) # <--- len() should be inside the sqrt()
>>> np.sum(abs(a_normalized)**2)/len(a_normalized) # average power of normalized points
4.000000000000001
```

dave-pi, 2021-07-16:
Regarding the 2nd question. It works for 8-PSK and for 16-QAM (although in the latter if you take the round() you will see that some elements have either a real of imaginary part, slightly above 1).

```py
In [44]: # 8-PSK
    ...: a = np.exp(2*np.pi*1j*(np.arange(8)/8))
    ...: average_amp = np.sum(abs(a))/len(a)
    ...: a_normalized = a/average_amp
    ...: np.sum(abs(a_normalized))/len(a_normalized) # average amp of normalized points
Out[44]: 1.0

In [45]: a_normalized.round(2)
Out[45]: 
array([ 1.  +0.j  ,  0.71+0.71j,  0.  +1.j  , -0.71+0.71j, -1.  +0.j  ,
       -0.71-0.71j, -0.  -1.j  ,  0.71-0.71j])

In [46]: # 16-QAM
    ...: a = np.array([
    ...:      -3+3j, -1+3j, +1+3j, +3+3j,
    ...:      -3+1j, -1+1j, +1+1j, +3+1j,
    ...:      -3-1j, -1-1j, +1-1j, +3-1j,
    ...:      -3-3j, -1-3j, +1-3j, +3-3j])
    ...: average_amp = np.sum(abs(a))/len(a)
    ...: a_normalized = a/average_amp
    ...: np.sum(abs(a_normalized))/len(a_normalized) # average amp of normalized points
Out[46]: 1.0

In [47]: a_normalized.round(2)
Out[47]: 
array([-1.  +1.j  , -0.33+1.j  ,  0.33+1.j  ,  1.  +1.j  , -1.  +0.33j,
       -0.33+0.33j,  0.33+0.33j,  1.  +0.33j, -1.  -0.33j, -0.33-0.33j,
        0.33-0.33j,  1.  -0.33j, -1.  -1.j  , -0.33-1.j  ,  0.33-1.j  ,
        1.  -1.j  ])
```

We do have some tests for this, although I'm not too familiar with them [qa_constellation.py](https://github.com/gnuradio/gnuradio/blob/fbf2082c0e110eed210e05894c356f3582397993/gr-digital/python/digital/qa_constellation.py) and [test_soft_decisions.py](https://github.com/gnuradio/gnuradio/blob/fbf2082c0e110eed210e05894c356f3582397993/gr-digital/python/digital/test_soft_decisions.py).

My suggestion is to create a PR for the first issue which is clearly wrong.

jfmadeira, 2021-07-19:
I agree with the solution to the first issue, thank you for opening the PR.

On the second issue, I think that is the most straightforward solution as well. Of note, even when using average amplitude scaling, it will not work for example at 64-QAM. The largest point will be +1.15+1.15j. If using average power scaling, then the maximum real and imag coordinates will be even greater.

PR #4909 (2021-09-07): gr-digital: Fix constellation normalization by average power
Correct power normalization calculation.
Add a qa test for amplitude and power normalization.

Partially fixes #4892
closed 2021-09-07 by mormj