gnuradio/gnuradio #4897: PDU to Tagged Stream fails with no attribute complex_t

mormj, 2021-07-16:
Any instantiation of a PDU to tagged stream fails in GRC - this is an example flowgraph.  
GR >= 3.9

[image]

```
Traceback (most recent call last):
  File "/share/gnuradio/gr310/src/gnuradio/build/asdf.py", line 146, in <module>
    main()
  File "/share/gnuradio/gr310/src/gnuradio/build/asdf.py", line 124, in main
    tb = top_block_cls()
  File "/share/gnuradio/gr310/src/gnuradio/build/asdf.py", line 83, in __init__
    self.blocks_tagged_stream_to_pdu_0 = blocks.tagged_stream_to_pdu(blocks.complex_t, 'packet_len')
AttributeError: module 'gnuradio.blocks' has no attribute 'complex_t'
```

solomonbstoner, 2021-07-16:
I got this error before. I think it is because of some difference between 3.9 and 3.10 perhaps?  Not sure. I never found the cause and it disappeared after I did a few things I cant rmb. 

What happens when you replace the PDU to Tagged Stream block with a new one?

solomonbstoner, 2021-07-18:
@mormj assign this to me. Ill fix it

solomonbstoner, 2021-07-18:
@mormj do you see two tagged streams to pdu in your search bar too? If I use the top one, I get the same error you get. But there's no error when I use the one under PDU Tools.
[image]

solomonbstoner, 2021-07-18:
If so, it is likely due to `/usr/local/share/gnuradio/grc/blocks/blocks_tagged_stream_to_pdu.block.yml` being leftover in the installed location after patch commit a8a9913136a64da903f190493bdc117b5349625e when said block was moved from blocks to pdu.
Just delete the installed `blocks_tagged_stream_to_pdu.block.yml` and you should be good to go

mormj, 2021-07-19:
That did it!!  Thanks @solomonbstoner

closed 2021-07-19 by mormj