gnuradio/gnuradio #4898: Proposal: Deprecate tag_checker

marcusmueller, 2021-07-16:
runtime has a class tag_checker, which sorts a tag vector upon construction, and then can be used to look up a tag with a given index. 

This functionality is only used in one block internally (`chunks_to_symbols`).

This functionality also isn't very useful:

1. Tags are already sorted when they come out of `get_tags_in_range`
2. There's many less awkward ways of extracting the elements from a sorted vector that have the same key; the containers literally have methods for that, and you can use the iterator from the last call to these to start your next search.

Now, this is public API, so, deprecate it in 3.9, remove in 3.10, or do the same +0.1.

mormj, 2021-07-16:
If the method doesn't really do anything when using current GR, I would think going ahead and removing it in 3.10 - deprecating in the next 3.9 release (even though they could coincide) would make sense.

dkozel, 2021-09-27:
@marcusmueller Is tag_checker removed on master now? If so this issue can be closed.

marcusmueller, 2021-09-27:
Yep, it is, @dkozel ! Closing this issue.

closed 2021-09-27 by marcusmueller