gnuradio/gnuradio #4914: symbol_sync_{ff,cc}: d_tags should be multiset, not vector

marcusmueller, 2021-07-17:
In tag propagation, the newly arrived tags are sorted into the internal tag container, `d_tags`. That's a lot of sorting to insert sorted elements into an explicitly sorted container!

If we used an implicitly sorted container like `std::multiset`, that effort would be reduced. We could then also just `multiset::erase(start, end)` to prune the "expired" tags from the internal books.

solomonbstoner, 2021-08-28:
I believe you're referring to this portion of the `collect_tags` function:
```C++
    get_tags_in_range(d_new_tags, 0, nitems_rd, nitems_rd + count);
    d_tags.insert(d_tags.end(), d_new_tags.begin(), d_new_tags.end());
    std::sort(d_tags.begin(), d_tags.end(), tag_t::offset_compare);
```
I have some doubts.
1. Is it neccessary to have `std::sort`? Based on the stream tags wiki, the tags offset are in absolute time, so why couldnt we just append the new tags in `d_new_tags` right at the end of the `d_tags` vector? Since the new tags will have larger tag offsets compared to the ones already sorted in `d_tags`, no sorting seems necessary then.
2. If `d_tags` is to be changed to a `multiset`, wouldnt that mean the implementation of functions like `get_tags_in_range` and all the gr-digital blocks which use `d_tags` need to also switch from vectors to multiset?

marcusmueller, 2024-01-29:
still desirable, not an API break.