gnuradio/gnuradio #4916: gr-iio uses localized functions to construct parameters for libiio

marcusmueller, 2021-07-17:
As #4886 shows, gr-iio fails with non-US locales, because it uses localized `to_string` to calculate parameter strings (the kernel sysfs and libiio interfaces take strings for numerical values).

A relatively elegant solution would be using `std::to_chars`; however, that requires C++17, and hence prohibits any backporting.

I think the easiest way in terms of coding would be to move away from the `vector<string>` interface of `set_params` towards an interface that takes `vector<keyvalue>`, where `keyvalue` is a `pair<string, variant<string, double, int, ??..>`, and then convert that variant to a string representation in `set_params`, instead of the caller of `set_params`.

willcode, 2021-07-17:
It seems we use std::to_string in quite a few places.
```
$ grep -rl std::to_string
gnuradio-runtime/lib/vmcircbuf_mmap_shm_open.cc
gnuradio-runtime/lib/block.cc
gnuradio-runtime/lib/flat_flowgraph.cc
gnuradio-runtime/lib/pmt/pmt_unv.cc
gnuradio-runtime/include/gnuradio/flowgraph.h
gnuradio-runtime/include/gnuradio/basic_block.h
gr-zeromq/lib/base_impl.cc
gr-iio/lib/fmcomms2_source_impl.cc
gr-iio/lib/fmcomms5_sink_impl.cc
gr-iio/lib/device_sink_impl.cc
gr-iio/lib/fmcomms2_sink_impl.cc
gr-iio/lib/fmcomms5_source_impl.cc
gr-uhd/lib/usrp_source_impl.cc
gr-uhd/lib/rfnoc_graph_impl.cc
gr-uhd/lib/usrp_sink_impl.cc
gr-audio/lib/alsa/alsa_impl.h
gr-soapy/python/soapy/bindings/soapy_types_python.cc
gr-soapy/lib/block_impl.cc
gr-soapy/lib/setting_string_conversion.h
gr-qtgui/grc/qtgui_waterfall_sink_x.block.yml
gr-qtgui/grc/qtgui_freq_sink_x.block.yml
gr-dtv/lib/atsc/atsc_fs_checker_impl.cc
gr-blocks/lib/wavfile_source_impl.cc
gr-blocks/lib/wavfile_sink_impl.cc
```

marcusmueller, 2021-07-18:
@willcode  Yep, and for a lot of these, I'm to blame. That's a mistake I did in the past, too: #4241 . Problem is that libiio uses these localized strings to exchange data with the kernel, which understandably doesn't understand how to parse a number with a comma in it.