gnuradio/gnuradio #4918: Metainfo test fails because of appstreamcli upstream bug

marcusmueller, 2021-07-18:
As reported in https://github.com/ximion/appstream/issues/331 , appstreamcli reports a deprecated property as error. 

The usage of an email address / mailto URL in "point of contact" is deprecated (which I think is a severe mistake and has been done based on the assumption that users might not have configured email).

Question is 

1. whether we let that tool¹ error out all it wants and declare our file valid, by bending our test case to ignore this specific error, or
2. whether we change our metainfo file for the worse, or 
3. whether we remove the check altogether, thereby never noticing if we should ever actually break it, but not running circles around appstreamcli.

I don't like any of the three options. I tend towards the first, but it's the most work.

----

¹  In my humble but very grumpy opinion, appstreamcli really doesn't have the greatest track record on being a good tool for validating their file format; they actually removed the formal specification of the archive-bound format they define, because it was hard to maintain. Now they have > 2000 lines of C code that validates the XML, with nothing like a schema, and in more than one case human-readable definition not matching the code

willcode, 2021-07-18:
This works:
`appstream-util validate-relax grc/scripts/freedesktop/org.gnuradio.grc.metainfo.xml`

marcusmueller, 2021-07-18:
@willcode you're a hero!

PR #4921 (2021-07-19): GRC: Test: Relax the checking of the metainfo file
This fixes a CI failure I introduced when appstreamcli's default checking started to reject mailto: contacts

Closes #4918
closed 2021-07-19 by marcusmueller