gnuradio/gnuradio #4937: gr-digital/examples/ofdm/tx_ofdm.grc example is broken.

drmpeg, 2021-07-20:
@dl1ksv Looks like commit https://github.com/gnuradio/gnuradio/commit/fbab6c383d6f86ab0de42a90a9182d22637a5b96 has a side effect. It breaks this flow graph in gr-digital.

[image]

dl1ksv, 2021-07-20:
@drmpeg  Worked for me. The error was not in fbab6c3 but in  98cadbf3d54176b474d8257ae0f2ae6f7c01fc3a.
But I see, you have already found it and fixed this issue

dl1ksv, 2021-07-22:
After I aplying #4936 the flowgraph still fails. The first time I probably tested a wrong branch. Sorry for that.
But now #4949 should fix this issue.

drmpeg, 2021-07-22:
Thanks for re-checking. I'll give https://github.com/gnuradio/gnuradio/pull/4949 a test.

PR #4949 (2021-08-07): grc: Sometimes the evaluation of a variable fails
The gr-digital/examples/tx_ofdm.grc fails to evaluate the
header_formater variable.

Fixes #4937
commit 1f1733eb (2021-08-07): grc: fix the evaluation of interdependent variables

The gr-digital/examples/tx_ofdm.grc fails to evaluate the
header_formater variable.

This pr fixes #4937
PR #4986 (2021-08-07): grc: fix the evaluation of interdependent variables (backport to maint-3.9)
The gr-digital/examples/tx_ofdm.grc fails to evaluate the
header_formater variable.

This pr fixes #4937

(cherry picked from commit 1f1733eb4489b48fb73509e7806df19e1c738092)

Backport https://github.com/gnuradio/gnuradio/pull/4949
commit ededb735 (2021-08-07): grc: fix the evaluation of interdependent variables

The gr-digital/examples/tx_ofdm.grc fails to evaluate the
header_formater variable.

This pr fixes #4937

(cherry picked from commit 1f1733eb4489b48fb73509e7806df19e1c738092)
closed 2021-08-07 by mormj