gnuradio/gnuradio #4941: GR3.8 Crashes upon placement of a QT-GUI block of any kind

esuter19, 2021-07-20:
Installed GR 3.8 via PyBOMBS along with gr-limesdr on Ubuntu 18.04 system.

When running GR with `pybombs -v run gnuradio` a few errors show up in the terminal:

```
<<< Welcome to GNU Radio Companion v3.8.3.1-10-g02f45ff9 >>>

Block paths:
	/home/ssrl/prefix-3.8/share/gnuradio/grc/blocks

Loading: "/home/ssrl/GNURadio/how_tf_do_i_get_this_to_work.grc"
>>> Done

Loading: "/home/ssrl/GNURadio/ares_lime_tx.grc"
>>> Done
Warning: restarting the docstring loader (crashed while loading 'qtgui_bercurve_sink')
Warning: restarting the docstring loader (crashed while loading 'qtgui_const_sink_x')
Warning: restarting the docstring loader (crashed while loading 'qtgui_edit_box_msg')
Warning: restarting the docstring loader (crashed while loading 'qtgui_freq_sink_x')
Warning: restarting the docstring loader (crashed while loading 'qtgui_histogram_sink_x')
Warning: docstring loader crashed too often
```
The program seems to run fine until I attempt to add a QT-GUI block to the flow graph. It then crashes without any further output to the terminal. If there was any indication of why it failed I would continue to debug on my own. I have made sure that the QT libraries are installed by running `sudo apt install libqwt-qt5-dev libqt5openglsdf5-dev python3-pyqt5` but that did not change anything. Another possible source of problems is that Anaconda is installed, but I have been running GR with it deactivated.

The verbose output is as follows:
```
ssrl@ssrlgnd:~/prefix-3.8/lib$ pybombs -v run gnuradio-companion
[DEBUG] Reading config info from file: /etc/pybombs/config.yml
[DEBUG] Parsing config file failed (/etc/pybombs/config.yml).
[DEBUG] Reading config info from file: /home/ssrl/.pybombs/config.yml
[DEBUG] New config items: ordereddict([('packagers', 'apt,pymod,pip,pkgconfig,cmd'), ('elevate_pre_args', ['sudo', '-H']), ('git-cache', '/home/ssrl/.pybombs/gitcache'), ('makewidth', 4), ('default_prefix', '/home/ssrl/prefix-3.8')])
[DEBUG] Template directory: /home/ssrl/.local/lib/python3.6/site-packages/pybombs/templates
[DEBUG] Inspecting config file: /home/ssrl/.pybombs/config.yml
[DEBUG] Using default_prefix as prefix (/home/ssrl/prefix-3.8)
[DEBUG] Choosing default prefix config dir: /home/ssrl/prefix-3.8/.pybombs
[DEBUG] Inspecting config file: /home/ssrl/prefix-3.8/.pybombs/config.yml
[DEBUG] Prefix source dir is: /home/ssrl/prefix-3.8/src
[DEBUG] Trying to load inventory file /home/ssrl/prefix-3.8/.pybombs/inventory.yml...
[DEBUG] Prefix-local recipe dir is: /home/ssrl/prefix-3.8/.pybombs/recipes
[DEBUG] Python version set by config file.
[DEBUG] Reading config info from file: /home/ssrl/prefix-3.8/.pybombs/config.yml
[DEBUG] New config items: {}
[DEBUG] Full list of recipe locations: ['/home/ssrl/prefix-3.8/.pybombs/recipes', '/home/ssrl/.pybombs/recipes/gr-etcetera', '/home/ssrl/.pybombs/recipes/gr-recipes', '/home/ssrl/.local/lib/python3.6/site-packages/pybombs/recipes']
[DEBUG] Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
[INFO] Prefix Python version is: 3.6.9
[INFO] PyBOMBS Version 2.3.4
[DEBUG] Adding recipe location: /home/ssrl/prefix-3.8/.pybombs/recipes
[DEBUG] Scanning directory '/home/ssrl/prefix-3.8/.pybombs/recipes' for recipes...
[DEBUG] Found 0 new recipes.
[DEBUG] Loading templates.
[DEBUG] Adding recipe location: /home/ssrl/.pybombs/recipes/gr-etcetera
[DEBUG] Scanning directory '/home/ssrl/.pybombs/recipes/gr-etcetera' for recipes...
[DEBUG] Found 126 new recipes.
[DEBUG] Loading templates.
[DEBUG] Adding recipe location: /home/ssrl/.pybombs/recipes/gr-recipes
[DEBUG] Scanning directory '/home/ssrl/.pybombs/recipes/gr-recipes' for recipes...
[DEBUG] Found 206 new recipes.
[DEBUG] Loading templates.
[DEBUG] Adding recipe location: /home/ssrl/.local/lib/python3.6/site-packages/pybombs/recipes
[DEBUG] Scanning directory '/home/ssrl/.local/lib/python3.6/site-packages/pybombs/recipes' for recipes...
[DEBUG] Found 11 new recipes.
[DEBUG] Loading templates.
[DEBUG] Initializing command class for command run
<<< Welcome to GNU Radio Companion v3.8.3.1-10-g02f45ff9 >>>

Block paths:
	/home/ssrl/prefix-3.8/share/gnuradio/grc/blocks

Loading: "/home/ssrl/GNURadio/how_tf_do_i_get_this_to_work.grc"
>>> Done

Loading: "/home/ssrl/GNURadio/ares_lime_tx.grc"
>>> Done
Warning: restarting the docstring loader (crashed while loading 'qtgui_bercurve_sink')
Warning: restarting the docstring loader (crashed while loading 'qtgui_const_sink_x')
Warning: restarting the docstring loader (crashed while loading 'qtgui_edit_box_msg')
Warning: restarting the docstring loader (crashed while loading 'qtgui_freq_sink_x')
Warning: restarting the docstring loader (crashed while loading 'qtgui_histogram_sink_x')
Warning: docstring loader crashed too often
ssrl@ssrlgnd:~/prefix-3.8/lib$
```

dkozel, 2021-07-21:
Hi @esuter19 Thanks for the report. We don't have a root cause for this error, it has occurred a small number of times in the past but the common factor isn't known.

The 3.9 release series does run on Ubuntu 18.04, but needs an updated pybombs11 and liblog4cpp to be installed. If you're comfortable trying those updates and installing 3.9 that would be a useful datapoint.

Since you have Anaconda setup, have you already tried installing GNU Radio through there?

velichkov, 2021-07-21:
Hi @esuter19,

Welcome to GNU Radio!

> The program seems to run fine until I attempt to add a QT-GUI block to the flow graph. It then crashes without any further output to the terminal. If there was any indication of why it failed I would continue to debug on my own.

To further debug this crash you will need to run `gnuradio-companion` under `gdb` and take a backtrace. Open a terminal and execute the following commands

```
source /home/ssrl/prefix-3.8/setup_env.sh
gdb -ex run --args python3 /home/ssrl/prefix-3.8/bin/gnuradio-companion
```
Then do whatever is necessary to crash it, go back in the terminal, now you should see the gdb's prompt - `(gdb)`, execute the following and provide the full output
```
backtrace
```

Alternatively as you are an Ubuntu user you may try taking a backtrace using [Apport](https://wiki.ubuntu.com/Apport)

esuter19, 2021-07-21:
Thank you both for the quick responses! 
@dkozel I have installed GNU Radio with Anaconda previously but it caused problems with the gr-limesdr OOT module that I need to use. It is still required for other applications, but for the time being I have been working with Anaconda deactivated. As of right now I am not sure which version of GNU Radio I will have to use for my application (waiting on another's decision) so I would like to stick with 3.8 since it is (relatively) stable and been out for a few years.

@velichkov I ran the commands you sent and now have a bit more information as to why it is crashing but no idea how to address the issues.
Output after `gdb -ex run --args python3 /home/ssrl/prefix-3.8/bin/gnuradio-companion`:
```
Starting program: /usr/bin/python3 /home/ssrl/prefix-3.8/bin/gnuradio-companion
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff0fda700 (LWP 29498)]
[New Thread 0x7fffe87d9700 (LWP 29499)]
[New Thread 0x7fffdffd8700 (LWP 29500)]
[Thread 0x7fffdffd8700 (LWP 29500) exited]
[Thread 0x7fffe87d9700 (LWP 29499) exited]
[Thread 0x7ffff0fda700 (LWP 29498) exited]
[New Thread 0x7fffdffd8700 (LWP 29556)]
[New Thread 0x7fffe87d9700 (LWP 29557)]
[New Thread 0x7ffff0fda700 (LWP 29558)]
[New Thread 0x7fffbb6f6700 (LWP 29564)]
[New Thread 0x7fffb3fff700 (LWP 29565)]
[Thread 0x7fffb3fff700 (LWP 29565) exited]
[New Thread 0x7fffb3fff700 (LWP 29567)]
<<< Welcome to GNU Radio Companion v3.8.3.1-10-g02f45ff9 >>>

Block paths:
	/home/ssrl/prefix-3.8/share/gnuradio/grc/blocks

Loading: "/home/ssrl/GNURadio/how_tf_do_i_get_this_to_work.grc"
[Thread 0x7fffbb6f6700 (LWP 29564) exited]
>>> Done

Loading: "/home/ssrl/GNURadio/ares_lime_tx.grc"
>>> Done
[Thread 0x7fffb3fff700 (LWP 29567) exited]
Warning: restarting the docstring loader (crashed while loading 'qtgui_bercurve_sink')
Warning: restarting the docstring loader (crashed while loading 'qtgui_const_sink_x')
Warning: restarting the docstring loader (crashed while loading 'qtgui_edit_box_msg')
Warning: restarting the docstring loader (crashed while loading 'qtgui_freq_sink_x')
Warning: restarting the docstring loader (crashed while loading 'qtgui_histogram_sink_x')
Warning: docstring loader crashed too often
[Thread 0x7ffff0fda700 (LWP 29558) exited]

Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
QList<QPrinter::PageSize>::QList (
    this=0x7fff9f7f3538 <QPrinterInfoPrivate::shared_null+24>)
    at ../../include/QtCore/../../src/corelib/tools/qlist.h:121
121	../../include/QtCore/../../src/corelib/tools/qlist.h: No such file or directory.
```
And after `backtrace`:
```
#0  0x00007fff9ecb3604 in QList<QPrinter::PageSize>::QList() (this=0x7fff9f7f3538 <QPrinterInfoPrivate::shared_null+24>)
    at ../../include/QtCore/../../src/corelib/tools/qlist.h:121
#1  0x00007fff9ecb3604 in QPrinterInfoPrivate::QPrinterInfoPrivate(QString const&) (name=..., this=Python Exception <class 'RecursionError'> maximum recursion depth exceeded while getting the str of an object: 
0x7fff9f7f3520 <QPrinterInfoPrivate::shared_null>)
    at painting/qprinterinfo_p.h:71
#2  0x00007fff9ecb3604 in __static_initialization_and_destruction_0 (__initialize_p=1, __priority=65535) at painting/qprinterinfo.cpp:35
#3  0x00007fff9ecb3604 in _GLOBAL__sub_I_qprinterinfo.cpp(void) ()
    at painting/qprinterinfo.cpp:163
```

velichkov, 2021-07-21:
Is this the full backtrace? It's a bit short so if there is something else please provide it as well. The crash seems to be during static initialization in some Qt library.

> I did notice that there is no src directory under /usr/share/qt4 which may be relevant.

The gnuradio 3.8 requires and should be built with Qt5 and not Qt4. If you have qt4 packages installed on your system and you don't need them for something else try removing them, especially the `libqt4-dev` and `libqwt-dev` packages.

See also #2506 and try executing the gnuradio's tests, go in the gnuradio's build directory and execute `make test`

```
cd /home/ssrl/prefix-3.8/src/gnuradio/build
make test
```

velichkov, 2021-07-21:
This problem looks similar to #3494 and #3215. Try removing the `qt-devel` package.

esuter19, 2021-07-22:
Removing the qt4 packages leftover from a GR 3.7 installation and reinstalling 3.8 via PyBOMBS solved the problem. Thank you for your patience!

closed 2021-07-22 by esuter19

dkozel, 2021-07-22:
Thanks for reporting back with what works! That'll help other folks.