gnuradio/gnuradio #4944: Some Python blocks may be able to be moved into C++

ncorgan, 2021-07-21:
There are a bunch of blocks (mostly hierarchical) developed in Python, I assume out of ease of development. This stops them from being used by C++ clients (or other hierarchical blocks). In theory, some of the could be reimplemented on the C++ layer. With GNU Radio's Python bindings (and maybe an interim compatibility `.py` file if necessary), these shouldn't break any Python clients, and the QA tests should still work.

Some of them use Python-specific stuff (socket stuff comes to mind) but certainly not all. These days, y'all seem to be more open to change than in the past, so I thought I'd bring it up.

mormj, 2021-07-21:
Can you start a checklist here of blocks that would be good candidates?

ncorgan, 2021-07-22:
I've gone through the first few modules and listed the technically viable blocks. I'll continue tomorrow.

ncorgan, 2021-07-23:
@mormj Done

solomonbstoner, 2021-07-23:
@ncorgan is the suggestion to create hier blocks in c++ to replace these hier blocks currently in python?

willcode, 2021-07-23:
Not totally related to this issue, but throwing it out there ...

When a block is used in a hier block, there will be only one downstream block. In-place operations are possible in many cases then, if blocks are written to support this. Separating the work "kernel" of a block from the `work()` of a block would allow multiple functions to be called from a hier block without the need to more threads, buffers, latency. This would not require (any? much?) scheduler work, just some changes to hier and participating blocks, and it would all be backward compatible.

ncorgan, 2021-07-23:
@solomonbstoner Correct. It should be possible to do this without breaking Python clients.

@willcode Ah, I wasn't aware of any non-block changes that would be needed. I assumed it would be a 1-to-1 conversion to some extent. That's obviously something to take into account for the necessary effort. FWIW, this doesn't apply to all Python conversions proposed above.

willcode, 2021-07-23:
@ncorgan I don't think there's any non-block work required for your idea (need to read all the blocks to know for sure). My comment was additional idea.

willcode, 2021-07-23:
@ncorgan Yah, there is a bunch of util stuff in python that would take some effort to port, and python versions would have to remain for backward compatibility. Could start with the easy/most valuable ones.

drmpeg, 2021-07-24:
For gr-dtv, those Python blocks could just be deleted. The ATSC receiver was completely reworked last year and doesn't use ```atsc_filter.py``` anymore. ```atsc_filter.py and ```atsc_rx.py``` are only used by ```file_atsc_rx.grc``` which could easily be deleted since it does the same thing as uhd_atsc_rx.grc (if you just add a file source block).

marcusmueller, 2021-07-30:
@drmpeg I'm hesitant to delete stuff abruptly, but would marking them "deprecated" on maint-3.9 and deleting them on master sound like a good idea?

drmpeg, 2021-07-30:
@marcusmueller Sure, I'm fine with that.

ncorgan, 2021-08-03:
WIP porting optfir (arbitrarily chosen): https://github.com/ncorgan/gnuradio/tree/optfir-to-cpp

willcode, 2021-08-03:
Backport note: `for_each` is C++-17.

ncorgan, 2021-08-03:
Ah, force of habit. Let me know if I need to defancy it, probably in the PR.

marcusmueller, 2021-08-04:
I, for one, think backporting might be a bit risky here.

willcode, 2021-08-04:
Definitely more work to backport. Up to @ncorgan whether it's worth it.