gnuradio/gnuradio #4952: Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.

ik8ozv, 2021-07-24:
Please Since i upgraded Ubuntu from version Ubuntu 20.10 to version 21.04, i have an issue with gnuradio running any GRC file.
i get the following warning:

<< Welcome to GNU Radio Companion v3.8.3.1-13-g6c7e7843 >>>

Block paths:
	/usr/local/share/gnuradio/grc/blocks

Loading: "/media/enzo/B038BD3C38BD0280/GRC/Sat_Lucky-7_IK8OZV.grc"
>>> Done

Generating: '/media/enzo/B038BD3C38BD0280/GRC/Lucky_7.py'

Executing: /usr/bin/python3 -u /media/enzo/B038BD3C38BD0280/GRC/Lucky_7.py

Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 2dd4
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffff

Please any Help to avoid the problem?
Thanks
Enzo

willcode, 2021-07-25:
That is just a warning. Is it causing any problems for you (other than the message)?

ik8ozv, 2021-07-25:
@willcode  It seems it does not cause problems but i get it on all files grc.

willcode, 2021-07-25:
Don't think we can do anything about this warning, not caused by GNU Radio, and causes no other problems, so closing. Please reopen if any side effects are found.

closed 2021-07-25 by willcode