gnuradio/gnuradio #4955: grc: cpp generation fails to generate vector variable

dl1ksv, 2021-07-26:
Define in grc a variable block with value
[float(x) for x in range(4)]

Python code will be generated but cpp code generation fails with:

Traceback (most recent call last):
  File "/usr/local/gnuradio/lib64/python3.9/site-packages/gnuradio/grc/gui/Application.py", line 720, in _handle_action
    generator.write()
  File "/usr/local/gnuradio/lib64/python3.9/site-packages/gnuradio/grc/core/generator/cpp_top_block.py", line 66, in write
    self._variable_types()
  File "/usr/local/gnuradio/lib64/python3.9/site-packages/gnuradio/grc/core/generator/cpp_top_block.py", line 301, in _variable_types
    var.format_expr(var_types[str(var.params['id'].value)])
  File "/usr/local/gnuradio/lib64/python3.9/site-packages/gnuradio/grc/core/blocks/block.py", line 479, in format_expr
    self.vtype = get_type(value, py_type)
  File "/usr/local/gnuradio/lib64/python3.9/site-packages/gnuradio/grc/core/blocks/block.py", line 460, in get_type
    return 'std::vector<' + get_type(str(evaluated[0]), type(evaluated[0])) + '>'
TypeError: 'NoneType' object is not subscriptable

That is why in in block.py
 ##
       def format_expr(self, py_type):
        """
        Evaluate the value of the variable block and decide its type.
        Returns:
            None
        """
        value = self.params['value'].value
        self.cpp_templates = copy.copy(self.orig_cpp_templates)

        # Determine the lvalue type
        def get_type(element, _vtype):
            evaluated = None
            try:
                evaluated = ast.literal_eval(element)`

evaluated = ast.literal_eval(element) fails for 
element = [float(x) for x in range(4)]

with ValueError: malformed node or string: <ast.ListComp object at 0x7fd8a0e92eb0>

solomonbstoner, 2021-08-28:
Ok. Assign that to me. Ill try to fix it.

solomonbstoner, 2021-08-28:
I believe the problem lies with `ast.literal_eval`. I got the same "malformed node or string" error with the code below
```python
import ast
ast.literal_eval("[float(x) for x in range(4)]")
```
This problem would be fixed by replacing `ast.literal_eval` with `eval`.
```python
eval("[float(x) for x in range(4)]")
```

This bug disappears once I replace `ast.literal_eval` with `eval` in lines 436 and 482.
```python
       # Determine the lvalue type
        def get_type(element, _vtype):
            evaluated = None
            try:
                evaluated = eval(element)
# ...
                if _vtype == None:
                    _vtype = type(evaluated)

        if self.vtype in ['bool', 'gr_complex'] or 'std::map' in self.vtype or 'std::vector' in self.vtype:
            evaluated = eval(value)
# ...
```

Not sure if its wise to change it though. `eval` not as safe as `ast.literal_eval` but the [latter "only considers a small subset of Python's syntax to be valid"](https://stackoverflow.com/questions/15197673/using-pythons-eval-vs-ast-literal-eval).

marcusmueller, 2024-01-29:
On main, with this fg:

[image]

https://gist.github.com/marcusmueller/c9741d435e8ec7aa5ba23db90a987334

we get a different error:

```text
Traceback (most recent call last):
  File "/home/marcus/.usrlocal/lib64/python3.11/site-packages/gnuradio/grc/gui/Application.py", line 758, in _handle_action
    generator.write()
  File "/home/marcus/.usrlocal/lib64/python3.11/site-packages/gnuradio/grc/core/generator/cpp_top_block.py", line 84, in write
    self._variable_types()
  File "/home/marcus/.usrlocal/lib64/python3.11/site-packages/gnuradio/grc/core/generator/cpp_top_block.py", line 346, in _variable_types
    var.format_expr(var_types[str(var.params['id'].value)])
  File "/home/marcus/.usrlocal/lib64/python3.11/site-packages/gnuradio/grc/core/blocks/block.py", line 475, in format_expr
    self.vtype = get_type(value, py_type)
                 ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/marcus/.usrlocal/lib64/python3.11/site-packages/gnuradio/grc/core/blocks/block.py", line 460, in get_type
    return f"std::vector<{get_type(str(evaluated[0]), type(evaluated[0]))}>"
                                       ~~~~~~~~~^^^
TypeError: 'NoneType' object is not subscriptable
```