gnuradio/gnuradio #4956: Build failure with Qwt 6.2

ryanvolz, 2021-07-26:
The recent Qwt 6.2.0 release has removed `qwt_compat.h` (see item #10 [here](https://sourceforge.net/p/qwt/git/ci/develop/tree/CHANGES-6.2)), resulting in build failures with all current branches. For example, [here's a conda build of 3.9.2](https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=356344&view=logs&jobId=d0d954b5-f111-5dc4-4d76-03b6c9d0cf7e&j=d0d954b5-f111-5dc4-4d76-03b6c9d0cf7e&t=841356e0-85bb-57d8-dbbc-852e683d1642). This is probably not an immediate problem given that the new Qwt version can be avoided for a little while, but it's certainly something to be aware of and to start planning for now.

wkazubski, 2021-07-30:
Causing build problems on openSUSE Factory and Tumbleweed

atsampson, 2021-08-19:
There's also the problem with Qwt 6.2 that `QwtRasterData::setInterval` has been removed - subclasses like GNU Radio's `TimeRasterData` now need to implement `interval` themselves.

bmuzika, 2021-09-01:
I've got a PR up that addresses this issue (#5034). Review would be greatly appreciated.

wkazubski, 2021-09-02:
I backported your commit to GR 3.8.3.1. The most important change was to make changes in gr-qtgui/lib/timeRasterGlobalData.cc and gr-qtgui/lib/waterfallGlobalData.cc conditional (only for Qwt-6.2), to mainain compatibility with older Qwt 6.1.x

wkazubski, 2021-09-03:
I did some tests and everything is working for me

marcoferrara10, 2021-10-30:
Hello everyone,
I've encountered the same issue on compiling gnuradio  and I've compiled successfully the git version with the head at the commit bcba1acbee1e5b6ded326b4962c1eb5dcf040e28 by applying the patch proposed by @wkazubski.

@wkazubski  thank you so much.

willcode, 2021-10-30:
I put in a draft PR https://github.com/gnuradio/gnuradio/pull/5282 against maint-3.8 using 0003-enables-use-of-qwt-6_2.patch.txt to see what CI thinks. We would need to get versions for master and maint-3.9 before we can merge it though.

willcode, 2021-10-30:
#5282 passed CI on 3.8, so trying #5283 against 3.9.

bmuzika, 2021-10-30:
I'm testing out some changes on this branch [1]. It is a modified version of my original patch

[1] https://github.com/bmuzika/gnuradio/tree/bugfix/fix-qwt-6.1-operation

willcode, 2021-10-30:
@bmuzika I made sure to credit you in the PRs and commits ... let me know if you have something you think should replace these, though. And I didn't put one in for master, so you can start there.

bmuzika, 2021-11-01:
From my testing, it seems that the only thing needed to make the original patch work is to restrict the implementation of `setInterval` to only be compiled for Qwt >= 6.2.

I tested out the changes in the branch linked above using an Ubuntu 20.04 VM and a Fedora 34 VM. A simple flowgraph of Signal Source -> Throttle -> QT GUI Sink worked on both. It also worked on my host system (openSUSE Tumbleweed, but I want to create a VM with a clean install to make sure it is actually working). I can make another PR, with these changes, if that these changes seem alright.

willcode, 2021-11-01:
@bmuzika Sure, put in a new PR on master. If `setInterval` was messing up the colormap for the waterfall, then it sounds reasonable.

bmuzika, 2021-11-03:
Here's the PR on master.

https://github.com/gnuradio/gnuradio/pull/5302

PR #5302 (2021-11-11): qt-gui: enables use of Qwt 6.2
# Pull Request Details

## Description

This PR enables support for Qwt 6.2 (without breaking support for older versions) and cleans up some uses of the Qwt5 compatibility header. 

## Related Issue

Fixes #4956

## Which blocks/areas does this affect?

qt-gui

## Testing Done