gnuradio/gnuradio #4962: gr-digital: Replace clock recovery MM with Symbol Sync?

solomonbstoner, 2021-07-28:
Since the Clock Recovery MM block is deprecated, should the [gfsk](https://github.com/gnuradio/gnuradio/blob/master/gr-digital/python/digital/gfsk.py) and [gmfsk](https://github.com/gnuradio/gnuradio/blob/master/gr-digital/python/digital/gmsk.py) demod hier blocks have their clock recovery MM replaced with the Symbol Sync block?

If so, shall the hier block parameters be changed to accept the parameters used by Symbol Sync block? (e.g. omega gain and mu gain to be replaced with loop bandwidth and damping factor as input) Or shall they remain unchanged (for backwards compatibility), meaning the hier block converts the parameters internally as shown in [slide 32](https://www.gnuradio.org/grcon/grcon17/presentations/symbol_clock_recovery_and_improved_symbol_synchronization_blocks/Andy-Walls-Samples-to-Digital-Symbols.pdf)?

willcode, 2021-07-28:
Yes, it should be replaced some time before the deprecated blocks are actually removed.

It will probably be necessary to do param translation in the existing hier blocks, keeping backward compatibility. There could be new hier blocks (in the existing files) that use the new params, and the old hier blocks would be deprecated.

solomonbstoner, 2021-07-28:
@willcode assign this issue to me. I wanna take the opportunity to learn about timing synchronization. Will take me a while but I think its a good starting point for me :) what do you say?

PR #4969 (2021-09-02): gr-digital: replace clk recovery MM in hier blks
Fixes #4962. However, I have no idea what kind of test would prove that its performance is the same as the Clock Recovery MM block.

As Clock Recovery MM is deprecated, this commit replaces them
in the hier blocks, gmfsk demod and gfsk demod, with Andy Wall's Symbol
Sync block. Backward compatibility is maintained by keeping the
parameters unchanged, and converting the two gain values to the
corresponding loop bandwidth value that Symbol Sync requires.

**Note:** This means the `mu` parameter is no longer needed. Should I leave it in or take it  [... cut]
closed 2021-09-02 by mormj