gnuradio/gnuradio #4964: Versioned wiki docs

jaredd, 2021-07-28:
The [GNURadio wiki ](https://wiki.gnuradio.org) is being positioned as the new documentation reference. GR is a rapidly evolving code base with features and functionality changing often. It seems that there will be several versions of GR in use at any one time, e.g. now 3.7, 3.8, 3.9, and 3.10 all seem to have some user activity. IDK how hard it is to version the wiki docs also, ala [CMake](https://cmake.org/cmake/help/v3.18/index.html) or [Python](https://docs.python.org/3/) docs. 

I imagine older versions (<= 3.8) won't get as much documentation, but as new versions are released I imagine the now current versions will have more complete docs.

dkozel, 2021-08-09:
@jaredd Agreed that this is a really desirable feature for the documentation. The documentation used to be versioned (at https://www.gnuradio.org/doc/ ) but very few people were contributing and the docs were both incomplete and out of date. Moving to the wiki has dramatically increased the amount of documentation contributed thanks to the much easier workflow, but fully separately versioning the docs on the wiki is difficult.

Both CMake and Python are using Sphinx to generate their versioned Docs, as GNU Radio used to. Their level of detail about contributing documentation changes is quite good.
https://devguide.python.org/docquality/

NeuroForLunch, 2021-08-13:
If there is any interest in having the Sphinx documentation on [Dash.app](https://kapeli.com/dash) for offline use I am willing to make that happen.