gnuradio/gnuradio #4966: interdependent objects in GRC stopped working reliably

marcusmueller, 2021-07-30:
Test case: packet_loopback_hier complains about an undefined `Const_PLD`, even if it's there; renaming and replacing does fix things at first, at some point this breaks again.

Must have been a change merged into master in the last month or so :( Very hard to bisect.

[image]

I'm really stumbling in the dark here. I'd be very thankful for any leads, @haakov or @skoslowski ?

(git 25c233e65 / master on Python 3.95 on F34)

haakov, 2021-07-30:
I'll take a look at it :smile:

drmpeg, 2021-07-30:
It's been fixed already. See https://github.com/gnuradio/gnuradio/issues/4937 and https://github.com/gnuradio/gnuradio/pull/4949

haakov, 2021-07-30:
Yep that seems to fix it for me too

mormj, 2021-08-07:
closed by #4949

closed 2021-08-07 by mormj