gnuradio/gnuradio #4968: Compilation error with Pybind 2.7.0

haakov, 2021-07-30:
Master branch, Arch Linux.

I recently upgraded from Pybind version 2.6.2 to 2.7.0, which caused this error to pop up when compiling GR:
```
[ 33%] Building CXX object gr-zeromq/python/zeromq/bindings/CMakeFiles/zeromq_python.dir/cmake_pch.hxx.gch
In file included from <command-line>:
/home/hkon/gnuradio/build/gr-zeromq/python/zeromq/bindings/CMakeFiles/zeromq_python.dir/cmake_pch.hxx:5:10: fatal error: /build/pybind11/src/pybind11-2.7.0/pybind11/include/pybind11/pybind11.h: No such file or directory
    5 | #include "/build/pybind11/src/pybind11-2.7.0/pybind11/include/pybind11/pybind11.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
This error seems to show up from all modules, gnuradio-runtime included. I believe it can be worked around by editing all the `cmake_pch.hxx` files manually, but reverting pybind to version 2.6.2 worked too.

willcode, 2021-07-30:
Note: Fedora 35 has v2.7.0.

haakov, 2021-08-04:
**Update:** The issue has been fixed in version 2.7.0-2. Arch Linux is now at 2.7.1-1.

Fix: https://github.com/pybind/pybind11/commit/7cc0ebb4753770ace452d118bb327cdd9e76dcbb

It looks like this fix has not reached Fedora 35 yet, leaving the issue open until it has.

schneider42, 2021-10-17:
According to https://src.fedoraproject.org/rpms/pybind11 Fedora is now also at pybind11-2.7.1-1.fc35