gnuradio/gnuradio #4976: Add gr-digital flow graphs to grc/tests/test_examples.py

drmpeg, 2021-08-02:
gr-digital has example flow graphs in sub-directories that aren't being tested by test_examples.py
They are:
```
gr-digital/examples/packet
gr-digital/examples/ofdm
gr-digital/examples/demod
gr-digital/examples/equalizers
```
@haakov

haakov, 2021-08-18:
Also some in gr-blocks

drmpeg, 2021-08-22:
I'd fix this myself if I knew how. My Python skills are too weak. When I add the following, it doesn't work. What am I doing wrong?
```
diff --git a/grc/tests/test_examples.py b/grc/tests/test_examples.py
index 2b590a4a9..2e6deb260 100644
--- a/grc/tests/test_examples.py
+++ b/grc/tests/test_examples.py
@@ -35,6 +35,11 @@ def gather_examples():
                 for pos_ex in os.scandir(path.join(file_dir, "examples/grc")):
                     if path.isfile(pos_ex) and pos_ex.name.endswith(".grc"):
                         example_paths.append(pos_ex)
+
+                # Some modules have their .grc files in a subdirectory called "ofdm"
+                for pos_ex in os.scandir(path.join(file_dir, "examples/ofdm")):
+                    if path.isfile(pos_ex) and pos_ex.name.endswith(".grc"):
+                        example_paths.append(pos_ex)
             except FileNotFoundError:
                 continue
     return example_paths
```

haakov, 2021-08-22:
Ah yeah that won't work because the line `for pos_ex in os.scandir(path.join(file_dir, "examples/grc")):` will throw an exception, and the script will carry on to the next module. Try adding a directory called `grc` in `gr-digital/examples/` :)

Should probably try to avoid hard-coding all the subdirectories though

sepiatone, 2021-11-23:
@drmpeg @haakov , the following should work. It also adds tests from gr-blocks (13) and gr-qtgui (2)

diff --git a/grc/tests/test_examples.py b/grc/tests/test_examples.py
index 2b590a4a9..063825e5d 100644
--- a/grc/tests/test_examples.py
+++ b/grc/tests/test_examples.py
@@ -30,11 +30,12 @@ def gather_examples():
                 for pos_ex in os.scandir(path.join(file_dir, "examples")):
                     if path.isfile(pos_ex) and pos_ex.name.endswith(".grc"):
                         example_paths.append(pos_ex)
-
-                # Some modules have their .grc files in a subdirectory called "grc"
-                for pos_ex in os.scandir(path.join(file_dir, "examples/grc")):
-                    if path.isfile(pos_ex) and pos_ex.name.endswith(".grc"):
-                        example_paths.append(pos_ex)
+
+                    if path.isdir(pos_ex):
+                        for pos_ex_2 in os.scandir(pos_ex):
+                            if path.isfile(pos_ex_2) and pos_ex_2.name.endswith(".grc"):
+                                example_paths.append(pos_ex_2)
+
             except FileNotFoundError:
                 continue
     return example_paths

sepiatone, 2021-11-23:
@drmpeg @haakov , the following should add the example flow graphs under gr-digital. It also adds some example flow graphs in sub-folders under gr-blocks (13) and gr-qtgui (2).

```

diff --git a/grc/tests/test_examples.py b/grc/tests/test_examples.py
index 2b590a4a9..063825e5d 100644
--- a/grc/tests/test_examples.py
+++ b/grc/tests/test_examples.py
@@ -30,11 +30,12 @@ def gather_examples():
                 for pos_ex in os.scandir(path.join(file_dir, "examples")):
                     if path.isfile(pos_ex) and pos_ex.name.endswith(".grc"):
                         example_paths.append(pos_ex)
-
-                # Some modules have their .grc files in a subdirectory called "grc"
-                for pos_ex in os.scandir(path.join(file_dir, "examples/grc")):
-                    if path.isfile(pos_ex) and pos_ex.name.endswith(".grc"):
-                        example_paths.append(pos_ex)
+
+                    if path.isdir(pos_ex):
+                        for pos_ex_2 in os.scandir(pos_ex):
+                            if path.isfile(pos_ex_2) and pos_ex_2.name.endswith(".grc"):
+                                example_paths.append(pos_ex_2)
+
             except FileNotFoundError:
                 continue
     return example_paths

```

drmpeg, 2021-11-23:
@sepiatone This has been taken care of in PR https://github.com/gnuradio/gnuradio/pull/5188

sepiatone, 2021-11-24:
> @sepiatone This has been taken care of in PR #5188

@drmpeg oh, I missed that!

closed 2022-06-17 by drmpeg