gnuradio/gnuradio #4978: Installing specific version (3.7.10) of gnuradio

Yash-dev1, 2021-08-03:
Hello, I am using Ubuntu version 16.04 currently. I am facing an issue installing a particular version (3.7.10) of gnu radio...Although I tried to follow the command for installing gnuradio 3.7, I always get installed the 3.7.9 version of it. But I want version 3.7.10 or greater for the gnuradio 3.7. Hence, I am really stuck here. I will be thankful for any leads.

willcode, 2021-08-03:
Please use chat.gnuradio.org or the mailing list for help requests. Github Issues is used for bug reports and feature requests.

closed 2021-08-03 by willcode

dkozel, 2021-08-03:
@Yash-dev1 You will have to install from source.
https://wiki.gnuradio.org/index.php/InstallingGR#From_Source

For any questions about doing so willcode is right, https://chat.gnuradio.org is the right place to ask. I hope you are successful.