gnuradio/gnuradio #4995: benchmark_nco.cc + benchmark_vco.cc

gvanem, 2021-08-09:
Trying to compile `gr-blocks/tests/benchmark_nco.cc` and `gr-blocks/tests/benchmark_vco.cc`, I got
errors on the use of `gr::nco<...>` and `gr::vco<...>`.

AFAICS, these should be `gr::blocks::nco<...>` and `gr::blocks::vco<...>`.
Hence my patches:
```diff
--- a/gr-blocks/tests/benchmark_nco.cc 2021-02-24 09:53:28
+++ b/gr-blocks/tests/benchmark_nco.cc 2021-08-09 13:44:31
@@ -17,12 +17,13 @@
 #include <gnuradio/nco.h>

 #ifdef HAVE_SYS_RESOURCE_H
-/* from man gtrusage
+/* from man getrusage
    "including <sys/time.h> is not required these days"
    So, we don't */
 #include <sys/resource.h>
 #endif

+#include <sys/time.h>
 #include <cstdio>
 #include <cstdlib>
 #include <cstring>
@@ -101,7 +102,7 @@

 void basic_sincos_vec(float* x, float* y)
 {
-    gr::blocks::nco<float, float> nco;
+    gr::nco<float, float> nco;

     nco.set_freq(2 * GR_M_PI / FREQ);

@@ -115,7 +116,7 @@

 void native_sincos_vec(float* x, float* y)
 {
-    gr::blocks::nco<float, float> nco;
+    gr::nco<float, float> nco;

     nco.set_freq(2 * GR_M_PI / FREQ);

@@ -126,7 +127,7 @@

 void fxpt_sincos_vec(float* x, float* y)
 {
-    gr::blocks::fxpt_nco nco;
+    gr::fxpt_nco nco;

     nco.set_freq(2 * GR_M_PI / FREQ);

@@ -139,7 +140,7 @@

 void native_sincos(float* x, float* y)
 {
-    gr::blocks::nco<float, float> nco;
+    gr::nco<float, float> nco;

     nco.set_freq(2 * GR_M_PI / FREQ);

@@ -151,7 +152,7 @@

 void fxpt_sincos(float* x, float* y)
 {
-    gr::blocks::fxpt_nco nco;
+    gr::fxpt_nco nco;

     nco.set_freq(2 * GR_M_PI / FREQ);

@@ -165,7 +166,7 @@

 void native_sin(float* x, float* y)
 {
-    gr::blocks::nco<float, float> nco;
+    gr::nco<float, float> nco;

     nco.set_freq(2 * GR_M_PI / FREQ);

@@ -177,7 +178,7 @@

 void fxpt_sin(float* x, float* y)
 {
-    gr::blocks::fxpt_nco nco;
+    gr::fxpt_nco nco;

     nco.set_freq(2 * GR_M_PI / FREQ);
```

And:
```diff
--- a/gr-blocks/tests/benchmark_vco.cc 2021-02-24 09:53:28
+++ b/gr-blocks/tests/benchmark_vco.cc 2021-08-09 13:29:56
@@ -24,6 +24,7 @@
 #include <sys/resource.h>
 #endif

+#include <sys/time.h>
 #include <cstdio>
 #include <cstdlib>
 #include <cstring>
@@ -124,7 +125,7 @@

 void native_vco(float* output, const float* input)
 {
-    gr::blocks::vco<float, float> vco;
+    gr::vco<float, float> vco;

     for (int j = 0; j < ITERATIONS / BLOCK_SIZE; j++) {
         vco.cos(output, input, BLOCK_SIZE, K, AMPLITUDE);
@@ -133,7 +134,7 @@

 void fxpt_vco(float* output, const float* input)
 {
-    gr::blocks::fxpt_vco vco;
+    gr::fxpt_vco vco;

     for (int j = 0; j < ITERATIONS / BLOCK_SIZE; j++) {
         vco.cos(output, input, BLOCK_SIZE, K, AMPLITUDE);
```

Adding a `#include <sys/time.h>` is needed for `timeval` (on MSVC at-least).

The output of `benchmark_nco.exe` seems okay to me:
```
          nop loop:  cpu:  0,000  steps/sec:        inf
       native sine:  cpu: 320000,000  steps/sec:  6,250e+01
         fxpt sine:  cpu: 85000,000  steps/sec:  2,353e+02
    native sin/cos:  cpu: 782000,000  steps/sec:  2,558e+01
      fxpt sin/cos:  cpu: 143000,000  steps/sec:  1,399e+02
 basic sin/cos vec:  cpu: 824000,000  steps/sec:  2,427e+01
native sin/cos vec:  cpu: 878000,000  steps/sec:  2,278e+01
  fxpt sin/cos vec:  cpu: 159000,000  steps/sec:  1,258e+02
```

willcode, 2021-08-09:
@gvanem Do you want to put in a pull request? If not, one of us can do that. What version(s) is this patch against?

gvanem, 2021-08-09:
> Do you want to put in a pull request? 

Sorry, I'm a *n00b* when it comes to git.

> What version(s) is this patch against?

From tip of *master* today.

willcode, 2021-08-09:
@gvanem No problem, this is straightforward. PR is in.

willcode, 2021-08-09:
@gvanem Did you build these tests manually? From the cmake files, it looks like it should be built but I don't see the binaries anywhere.

gvanem, 2021-08-09:
> Did you build these tests manually?

Yes. Manually. No via CMake.

closed 2021-08-23 by mormj