gnuradio/gnuradio #4998: GRC: Can't use sc16 format with USRP

mchlwntz, 2021-08-09:
In the past we could use a file source with format "short" and vector length of 2 to send complex int16 to the USRP. Now this produces an error that the source IO type doesn't match the sink IO type. It seems like this was introduced by this commit, which checks the types in addition to the sizes: https://github.com/gnuradio/gnuradio/commit/d44ee73deeaab86a7cc5820ef62688f291331b88#diff-07eb034ef7119f3142143792b6a91429ea645e3dd21cd48251d5332abf7486a2. Maybe there needs to be a special case for sc16 and the other hardware formats.

willcode, 2021-08-09:
The type checking got a little stricter. Previously anything with the same number of bytes per item would match. Unfortunately, there isn't a short-to-ishort block. Maybe there should be one.

Making a vector of 2 shorts match an ishort is a possibility.

You can still feed complex values to the USRP Sink and have it send 16-bit (or smaller) values over the wire, if you're mainly looking for a more efficient OTW format.

willcode, 2021-08-09:
There's a missing alias in `grc/core/Constants.py`. I'll put in a fix.

dl1ksv, 2021-08-11:
@willcode I have only limited means to test #4999, but as far as I could test: it works

willcode, 2021-08-11:
Worked for C2I -> I2C, but not for USRP -> I2C
`ValueError: itemsize mismatch: usrp_source0:0 using 4, interleaved_short_to_complex0:0 using 2`

ryanvolz, 2021-08-14:
Hey, it's one of those times when you go to file a bug and find that some else recently did it!

This also kinda/sorta applies to the interleaved char blocks; I have an OOT block than uses an "sc8" type which previously worked with the output of the complex->ichar block, but now won't. I don't know if any other blocks out there use "sc8", but it would sure be nice if similar changes from #5001 could also be applied to the interleaved char blocks.

willcode, 2021-08-14:
It's one of those times where you were wondering if the other type option needed the same kind of change and ...

OK, I'll check it out.

willcode, 2021-08-14:
@mchlwntz Does the fix we're talking about work for you? The only downside of the I2C/C2I and I2B/B2I adapter blocks is that they do an unnecessary memcpy (to be addressed some day).

mchlwntz, 2021-08-16:
I can't test this at the moment, but if I understand correctly, the fix involves: (complex file source) --> (complex to ishort) --> (USRP sink)? If so, I don't see how that addresses the problem of using files that are already saved in the sc16 format? I would need to convert them all to complex floats which is twice the original size. The complex to ishort block also appears to scale every sample, which seems like a downside.

Also wondering if other blocks can interface with the USRP in this format, or if we need to always go through the complex to ishort? I've got plenty of flowgraphs to do fixed point processing where it'd be a pain to go back to complex floats just to get to the USRP, but maybe those just need the same fix as in the proposed YAML. 

Let me know if I misunderstood anything. Thanks!

willcode, 2021-08-16:
Then we may need either (1) a "cast" block which would incur a memcpy but otherwise do nothing, or (2) a mod to GRC to tell it that sc16 and short[2] are compatible. It could be that sc16 is just an alias for short[2] and should be shown as short[2] in the flowgraph.

#5001 is still useful, but as you say it does not address playback of a short[2] stream from a file. @haakov is there an "easy" way  to do this. The only user-visible effect would be that there would be no distinct color for sc16 I think.

ryanvolz, 2021-08-16:
Most (all?) processing blocks can't handle sc16 or sc8 types, so I think fixing the conversion sc->complex/complex->sc blocks should cover most uses besides the file source/sink. Thus, I think the best solution to that particular problem would be to add sc16 (and maybe complex versions of all the existing types?) as additional output/input types for the file blocks.

haakov, 2021-08-18:
Hm, tricky. But yeah I suppose the easiest way now that we have https://github.com/gnuradio/gnuradio/pull/5001 is to add types to the file blocks like @ryanvolz suggested. However, adding "interchangeable" type support to GRC (which shouldn't be too much work either) feels like more of a long-term fix to me

willcode, 2021-08-19:
Choices seem to be:
- Make sc16/sc8 first class types, like complex, and let them grow into other blocks as needed
- Change GRC to see sc16 == short[2], sc8 == byte[2]
- Change uhd/soapy blocks to use vectors and forget about sc16/sc8 altogether, remove types from GRC

PR #5127 (2021-10-05): grc: Add aliases for "sc16==short" and "sc8==byte"
When type checking was added for GRC connections, the valid connections between a block with the "sc16" type (e.g. UHD blocks) and a block with the "short" type and a vector length of 2 (e.g. ishort->complex) now fail. Adding these aliases fixes that as a stop-gap until a better solution can be found.

Fixes #4998. See further discussion on this subject in #5001.
closed 2021-10-05 by mormj