gnuradio/gnuradio #5002: Feature Request: Integration with Tensorflow or Torch

NeuroForLunch, 2021-08-13:
A lot of signal finding and decoding depends on manually designed filters but if there was an addition of Torch or Tensorflow, then pipelines could be constructed which find signals and decode them using machine learning techniques. If someone is willing to help with this I am willing to add to the effort.

willcode, 2021-08-13:
This would be an OOT module. It sounds like a great work area, so we can leave this Issue up for a bit to get any other responses, but it won't be a core feature.

marcusmueller, 2021-08-16:
gr-dnn exists, see "Deep learning inference in GNU Radio with ONNX"; some other implementations exist, too.

NeuroForLunch, 2021-08-16:
Thanks for the info - I just looked at gr-dnn and it seems it has been abandoned due to lack of functionality and gr-wavelearner doesn't seem very useable for general applications.  I was hoping to see some empirical applications but haven't found any yet.